hadess/iio-sensor-proxy #6: Lenovo Yoga Pro 3

andreasn, 2015-01-06:
Tried to get it running on a Lenovo Yoga Pro 3

Here is the debug data I got by running sudo G_MESSAGES_DEBUG=all /usr/local/sbin/iio-sensor-pxy:

** (process:4806): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200073.0.auto/iio:device3
** (process:4806): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200073.0.auto/iio:device3 at IIO Buffer accelerometer
** (process:4806): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200073.0.auto/trigger3
** (process:4806): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:4806): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:4806): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:4806): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:4806): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:4806): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available
** (process:4806): DEBUG: No new data available

hadess, 2015-01-06:
Which kernel are you using? Is that after a clean boot, or did you suspend/resume your laptop before trying this? If you suspended it, could you please try from a clean boot?

also what's the output of:
```
ls /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200073.0.auto/iio:device3/scan_elements/
```

andreasn, 2015-01-07:
kernel is 3.17.7-300.fc21.x86_64

Yes, fresh after reboot

sudo ls /sys/devices/pci0000\:00/0000\:00\:14.0/usb1/1-8/1-8\:1.0/0003\:048D\:8386.0001/HID-SENSOR-200073.0.auto/iio\:device5/scan_elements/
in_accel_x_en	  in_accel_y_en     in_accel_z_en
in_accel_x_index  in_accel_y_index  in_accel_z_index
in_accel_x_type   in_accel_y_type   in_accel_z_type

ai, 2015-01-19:
This issuee is very important to mee too (same Yoga 3 Pro). Maybe I can help with some data?

pfulop, 2015-01-22:
Same with thinkpad-yoga 

E: ABS=7
E: DEVPATH=/devices/virtual/input/input19
E: EV=9
E: ID_INPUT=1
E: ID_INPUT_ACCELEROMETER=1
E: ID_INPUT_ACCELEROMETER_ORIENTATION=undefined
E: MODALIAS=input:b0006v0001p0002e0000-e0,3,kra0,1,2,mlsfw
E: NAME="IIO Accelerometer Proxy"
E: PRODUCT=6/1/2/0
E: PROP=0
E: SUBSYSTEM=input
E: TAGS=:seat:
E: USEC_INITIALIZED=625778

andreasn, 2015-01-23:
@hadess I'll be bringing this device to FOSDEM if you want to test it out yourself. I'll bring rawhide on a USB stick to make sure it's the very latest kernel and stuff.

andreasn, 2015-02-18:
Crazy enough, this worked briefly the other day, but never again after a reboot...
Will see if I'm able to get it running again somehow.

elad661, 2015-03-23:
I think this is the same as issue #1

@andreasn can you please try using a newer kernel, and also the hysteresis workaround listed in issue #1?

I suspect the twist and the yoga use the same sensor board. If you can't test with a newer kernel, can you please run `lsusb | grep 0483:91d1`? this is the USB ID of the sensor board on the thinkpad twist. If the thinkpad yoga 3 has a device with the same USB ID, it's probably the same sensor board and thus affected by the same driver bugs.

andreasn, 2015-04-02:
lsusb | grep 0483:91d1
turns up blank here

sils1297, 2015-07-08:
Hi, I get the same output (see http://pastebin.com/KD9MD9pu for mine) with

    $ uname -a
    Linux lsant 4.1.1-1-ARCH #1 SMP PREEMPT Sat Jul 4 13:10:39 UTC 2015 x86_64 GNU/Linux

using gnome 3.17.3 on ArchLinux with testing and gnome-devel active.

    $ lsusb | grep 0483:91d1
    $ # No output for me too!

elad661, 2015-07-08:
Yeah that means it's a different kind of chip with a different kind of problem. The best way to figure out why it's not working would be sending an email to the linux-iio mailing list and asking for their help, as this sounds like a kernel driver bug or a firmware bug.

hadess, 2015-07-22:
As mentioned by @elad661 above, best communicate this problem with the linux-iio mailing-list. Make sure to test with the latest kernels though!

closed 2015-07-22 by hadess

elad661, 2015-08-11:
I know this is closed, but:

For everyone who have this laptop, use the latest kernel (such as the one in Fedora Rawhide) and the latest iio-sensor-proxy.

We tested it today in GUADEC on @andreasn's machine, the machine has an accelerometer, ambient light sensor, and a compass.