hadess/iio-sensor-proxy #16: Yoga 2 Pro

johnpaul5, 2015-04-04:
Hi, trying to get this running on a Lenovo Yoga 2 Pro (ubuntu gnome, kernel 3.16), it looks like it detects the orientation changes ('Emitted orientation changed') but the screen is not rotating. Here's the debugging results : 

    ** (process:5326): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0003/HID-SENSOR-200041.7.auto/iio:device5
    ** (process:5326): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0003/HID-SENSOR-200041.7.auto/iio:device5 of type ambient light sensor at IIO Buffer Light sensor
    ** (process:5326): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0003/HID-SENSOR-200073.0.auto/iio:device0
    ** (process:5326): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0003/HID-SENSOR-200073.0.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
    ** (process:5326): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0003/HID-SENSOR-200073.0.auto/trigger0
    ** (process:5326): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
    ** (process:5326): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
    ** (process:5326): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
    ** (process:5326): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
    ** (process:5326): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
    ** (process:5326): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
    ** (process:5326): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0003/HID-SENSOR-200041.7.auto/trigger5
    ** (process:5326): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:5326): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:5326): DEBUG: Read from IIO: 82, -802, -353
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): -82, 802, -353
    ** (process:5326): DEBUG: Emitted orientation changed: from undefined to normal
    ** (process:5326): DEBUG: Read from IIO: 195
    ** (process:5326): DEBUG: Light level sent by driver (quirk applied): 195.000000 (unit: lux)
    ** (process:5326): DEBUG: Emitted light changed: from 0.000000 to 195.000000
    ** (process:5326): DEBUG: Read from IIO: 90, -809, -425
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): -90, 809, -425
    ** (process:5326): DEBUG: No new data available
    ** (process:5326): DEBUG: Read from IIO: -1164, 208, -222
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): 1164, -208, -222
    ** (process:5326): DEBUG: Emitted orientation changed: from normal to right-up
    ** (process:5326): DEBUG: Read from IIO: 156
    ** (process:5326): DEBUG: Light level sent by driver (quirk applied): 156.000000 (unit: lux)
    ** (process:5326): DEBUG: Emitted light changed: from 195.000000 to 156.000000
    ** (process:5326): DEBUG: Read from IIO: -950, 123, -274
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): 950, -123, -274
    ** (process:5326): DEBUG: No new data available
    ** (process:5326): DEBUG: Read from IIO: -978, 176, -219
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): 978, -176, -219
    ** (process:5326): DEBUG: No new data available
    ** (process:5326): DEBUG: Read from IIO: -982, 166, -224
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): 982, -166, -224
    ** (process:5326): DEBUG: No new data available
    ** (process:5326): DEBUG: Read from IIO: 93, -874, -177
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): -93, 874, -177
    ** (process:5326): DEBUG: Emitted orientation changed: from right-up to normal
    ** (process:5326): DEBUG: No new data available
    ** (process:5326): DEBUG: Read from IIO: 159, -824, -292
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): -159, 824, -292
    ** (process:5326): DEBUG: Read from IIO: 195
    ** (process:5326): DEBUG: Light level sent by driver (quirk applied): 195.000000 (unit: lux)
    ** (process:5326): DEBUG: Emitted light changed: from 156.000000 to 195.000000
    ** (process:5326): DEBUG: Read from IIO: 102, -806, -371
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): -102, 806, -371
    ** (process:5326): DEBUG: No new data available
    ** (process:5326): DEBUG: Read from IIO: 112, -746, -270
    ** (process:5326): DEBUG: Accel sent by driver (quirk applied): -112, 746, -270
    ** (process:5326): DEBUG: No new data available

Thanks for any help

hadess, 2015-04-04:
Use the version tagged 0.1 if you want any integration:
https://github.com/hadess/iio-sensor-proxy/releases

Newer versions require changes to user-space which aren't upstreamed yet.

closed 2015-04-04 by hadess

johnpaul5, 2015-04-05:
version 0.1 works like a charm, thank you so much for this script!