hadess/iio-sensor-proxy #19: Operation with Fedora 22...

ratboy666, 2015-05-12:
I have an Acer W700 tablet, with Fedora 22 (beta). Gnome 3.16.1, Kernel 4.0.2. I built and installed iio-sensor-proxy 0.1. The tablet screen doesn't rotate. So, I took a debug dump (below) which shows that the rotation is being detected -- but the screen still not rotating. Verified that the daemon process is running after startup:
[root@dejah lengine]# systemctl status iio-sensor-proxy.service
● iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/usr/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: disabled)
   Active: active (running) since Tue 2015-05-12 07:13:37 EDT; 9h ago
 Main PID: 775 (iio-sensor-prox)
   CGroup: /system.slice/iio-sensor-proxy.service
           └─775 /usr/local/sbin/iio-sensor-proxy

May 12 07:13:37 dejah systemd[1]: Started IIO Sensor Proxy service.
May 12 07:13:37 dejah systemd[1]: Starting IIO Sensor Proxy service...
[root@dejah lengine]# 

And now the debug dump (started with tablet in "normal" orientation, rotated so right edge up, then rotated back to "normal"):

[root@dejah iio-sensor-proxy-0.1]# G_MESSAGES_DEBUG=all /usr/local/sbin/iio-sensor-proxy
** (process:2856): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:8087:0A04.0001/HID-SENSOR-200073.0.auto/iio:device4
** (process:2856): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:8087:0A04.0001/HID-SENSOR-200073.0.auto/iio:device4 at IIO Buffer accelerometer
** (process:2856): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:8087:0A04.0001/HID-SENSOR-200073.0.auto/trigger4
** (process:2856): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2856): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2856): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2856): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2856): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2856): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2856): DEBUG: Read from IIO: -12, -248, -79
** (process:2856): DEBUG: Sent by driver (quirk applied): 12, 248, -79
** (process:2856): DEBUG: Emitted orientation changed: from undefined to normal
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: Read from IIO: -46, -248, -33
** (process:2856): DEBUG: Sent by driver (quirk applied): 46, 248, -33
** (process:2856): DEBUG: Read from IIO: -209, -120, -76
** (process:2856): DEBUG: Sent by driver (quirk applied): 209, 120, -76
** (process:2856): DEBUG: Emitted orientation changed: from normal to right-up
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: Read from IIO: -230, -33, -92
** (process:2856): DEBUG: Sent by driver (quirk applied): 230, 33, -92
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: Read from IIO: -163, -166, -99
** (process:2856): DEBUG: Sent by driver (quirk applied): 163, 166, -99
** (process:2856): DEBUG: Read from IIO: -23, -243, -69
** (process:2856): DEBUG: Sent by driver (quirk applied): 23, 243, -69
** (process:2856): DEBUG: Emitted orientation changed: from right-up to normal
** (process:2856): DEBUG: Read from IIO: -7, -240, -58
** (process:2856): DEBUG: Sent by driver (quirk applied): 7, 240, -58
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: Read from IIO: -12, -240, -97
** (process:2856): DEBUG: Sent by driver (quirk applied): 12, 240, -97
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: Read from IIO: -15, -248, -69
** (process:2856): DEBUG: Sent by driver (quirk applied): 15, 248, -69
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
** (process:2856): DEBUG: No new data available
^C

hadess, 2015-05-13:
So the output in the debug is correct?

What's the output of
```udevadm monitor --udev```
when reproducing the problem?

And is gnome-settings-daemon running in the session (I assume a plain GNOME setup?).

ratboy666, 2015-05-13:
Thanks a whole bunch for having a look! Automatic rotation and ambient light sensing are the two things still missing from my "tablet experience", and I am really looking forward to having these working!

Yes, the debug output is accurate -- I ran it and copy/pasted from the terminal.

Yes, this is a "plain" gnome 3 -- stock fedora 22 (so, I presume "plain" -- not sure precisely what that really is -- I do use a handful of gnome shell extensions: applications menu, battery percentage, battery power statistics shortcut, caffeine, clipboard indicator, coverflow alt-tab,  project hamster extension,  refresh wifi  connections,  settingscenter, suspend button,  topicons, transparent top bar and window overlay icons).

gnome-setting-daemon is running: as pid 1583 (see below). Note that iio-sensor-proxy is running as pid 762 (I gathered both pids) . Is there a dependency there?

$ ps ax | grep gnome
 1140 tty1     Ssl+   0:00 /usr/libexec/gdm-x-session --run-script gnome-session
 1288 tty1     Sl+    0:00 gnome-session
 1546 tty1     Sl+    0:00 /usr/libexec/at-spi2-registryd --use-gnome-session
 1579 tty1     Sl+    0:00 /usr/bin/gnome-keyring-daemon --start --components=pkcs11
 1583 tty1     Sl+    0:02 /usr/libexec/gnome-settings-daemon
 1656 tty1     Sl+    8:10 /usr/bin/gnome-shell
 1732 tty1     Sl+    0:00 /usr/libexec/gnome-shell-calendar-server
 1888 tty1     Sl+    0:00 /usr/libexec/gnome-user-share-obexpush
 1900 tty1     Sl+    0:00 /usr/bin/gnome-software --gapplication-service
 2100 tty1     Sl+    0:00 /usr/libexec/evolution-calendar-factory-subprocess --bus-name org.gnome.evolution.dataserver.Subprocess.Backend.Calendarx2042x2 --own-path /org/gnome/evolution/dataserver/Subprocess/Backend/Calendar/2042/2
 2119 tty1     Sl+    0:00 /usr/libexec/evolution-calendar-factory-subprocess --bus-name org.gnome.evolution.dataserver.Subprocess.Backend.Calendarx2042x3 --own-path /org/gnome/evolution/dataserver/Subprocess/Backend/Calendar/2042/3
 2154 tty1     Sl+    0:00 /usr/libexec/evolution-addressbook-factory-subprocess --bus-name org.gnome.evolution.dataserver.Subprocess.Backend.AddressBookx2120x2 --own-path /org/gnome/evolution/dataserver/Subprocess/Backend/AddressBook/2120/2
 3729 tty1     Sl+    0:01 /usr/libexec/gnome-terminal-server
 3732 tty1     S+     0:00 gnome-pty-helper
 4758 pts/4    S+     0:00 grep --color=auto gnome

$ ps ax | grep iio
  762 ?        Ss     0:01 /usr/local/sbin/iio-sensor-proxy
 4877 pts/4    S+     0:00 grep --color=auto iio

Now the udevadm output:

$ sudo udevadm monitor --udev
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing

UDEV  [6561.064666] change   /devices/virtual/input/input6 (input)
UDEV  [6566.674530] change   /devices/virtual/input/input6 (input)

I stated udevadm, rotated the table to right edge up, and then back to "normal" orientation.

hadess, 2015-05-13:
> I have an Acer W700 tablet, with Fedora 22 (beta). Gnome 3.16.1, Kernel 4.0.2. I built and installed iio-sensor-proxy 0.1. 

Let me guess. You say you're using version 0.1, but not the 0.1 tag.

Use the 0.1 tag if you want integration in GNOME (or anything else for that matter). There are a couple of bugs to fix before GNOME is getting integrated, see the other issues filed.

closed 2015-05-13 by hadess

ratboy666, 2015-05-13:
@hadess 

I tried with the current head, and with tag 0.1 (the release tag). But -- I can see that the daemon is running (and I even think the ambient light sensor is being supported with the current head!).

This is exciting (this tablet is the best thing yet for my needs). So, I guess I will wait on the gnome fixes, and keep an eye on this project. Thanks for your help!

Fred.