hadess/iio-sensor-proxy #22: Limit which applications can get sensor data

hadess, 2015-05-20:
Only foreground applications? Only to applications in the current seat? Should applications request it through an agent a-la Geoclue?

joshtriplett, 2015-05-22:
There are two independent concerns that seem applicable here.

First, there's the normal restriction to applications in the current seat, just as with local input devices or audio devices.  I would suggest handling this via the standard "uaccess" mechanism.

Second, even for such applications that qualify, sufficiently clever applications could infer keyboard or mouse events from high-precision accelerometer information.  For sandboxed applications that intentionally do not have access to input devices when not in the foreground, that could be a concern; however, for non-sandboxed applications, I don't think this is worth solving.  So, I would suggest documenting that access to raw accelerometer data (rather than heavily cooked data like orientation in 90 degree increments) can potentially leak information about user input ,and then letting sandboxing mechanisms handle this via their permissions mechanism.

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/22

closed 2020-01-09 by hadess