hadess/iio-sensor-proxy #25: MacBook Pro <Early 2011>

miguxbe, 2015-05-24:
Testing with my 'old' MacBook. Theoretically it detects devices right, but does not appear the "IIO Accelerometer Proxy" udev after run it $ udevadm info --export-db.

The debug output:

```bash
$ G_MESSAGES_DEBUG=all /usr/local/sbin/iio-sensor-proxy    
** (process:3476): DEBUG: Found HWMon light at /sys/devices/platform/applesmc.768
** (process:3476): DEBUG: Found device /sys/devices/platform/applesmc.768 of type ambient light sensor at Platform HWMon Light
** (process:3476): DEBUG: iio-sensor-proxy is already running
```

miguxbe, 2015-05-24:
It's looks like that this patch does not solve the problem, at least for me. The only thing I found was this after recompiling and restarting the service.

``` bash
P: /devices/virtual/hwmon/hwmon0
E: DEVPATH=/devices/virtual/hwmon/hwmon0
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=319640
```
And after run iio-sensor-prxy with DEBUG flag it returns the same massages.

hadess, 2015-05-24:
What's the output of:
```
systemctl status iio-sensor-proxy.service
```

On your system?

hadess, 2015-05-24:
FWIW, it works on the Mac I tested it on. You'll need to run ``` monitor-sensor ``` to see changes in the readings, and I ran ```./autogen.sh --prefix=/usr --sysconfdir=/etc``` before compiling and installing.

miguxbe, 2015-05-24:
Ok, the last tip was the key, now the service is active :+1 and the monitor-sensors returns values. Thanks. I was trying to reload again and again the service but was impossible to get it active (running).

commit 5436d07f (2015-05-24): rules: Add missing systemd TAG to autostart daemon

The daemon wasn't starting on, at least, MacBooks because the systemd
tag was missing, not starting the systemd iio-sensor-proxy.service unit.

Closes #25
closed 2015-05-24 by miguxbe