hadess/iio-sensor-proxy #28: Remove hacky scaling in Compass driver

hadess, 2015-06-03:
As soon as the kernel driver has been fixed.

hadess, 2015-06-03:
@bentiss Do you have a reference on that fix?

bentiss, 2015-06-03:
I spent some time today trying to understand if the iio devices exported the unit and exponents.
It occurs that they should already export the correct scale (as in what is done with accel):
For instance, on the ALS, I have:

    $ cat /sys/bus/iio/devices/iio:device0/in_intensity_scale
    0.000000010

(the device is declared with a unit exponent of -8).
There is one problem though. The current kernel code does not work with exponents greater than 6 because they are using a micro precision. This can be changed to nano quite easily and make the "scale" file work with up to +-9. That also means that a scale of 0 means that there is no scale available. In this case, we should probably raise a kernel bug to fix the report descriptor to have a proper scale.

The various scales are given in drivers/iio/common/hid-sensors/hid-sensor-attributes.c . According to this file, the compass should report a unit in gauss, once the scale factor is applied.

hadess, 2015-06-04:
> According to this file, the compass should report a unit in gauss, once the scale factor is applied.

According to the hid-sensors-usages.docx from MS, what we're reading should be in degrees (SENSOR_DATA_TYPE_MAGNETIC_HEADING_COMPENSATED_MAGNETIC_NORTH_DEGREES).

commit f04fe60a (2015-07-01): main: Apply scale from readings for IIO buffered drivers

Add a scale output parameter in process_scan_1 so that we pass scale
information from each driver read to the driver, and apply the scale
before passing it on to the main daemon.

Closes #28
closed 2015-07-01 by hadess