hadess/iio-sensor-proxy #29: Limit access to heading to Geoclue

hadess, 2015-06-03:
It's not directly useful to applications, as it's exporting the magnetic North heading, not the True North heading. Geoclue will usually have enough information to convert it.

closed 2015-06-03 by hadess

volkov0aa, 2015-06-08:
I want to write the plugin for Qt Sensors and QCompassReading documentation states: "The compass returns the azimuth of the device as degrees from magnetic north". So it can be useful in Qt applications.

hadess, 2015-06-08:
Qt Sensors and QCompassReading should access it through Geoclue, not directly.

volkov0aa, 2015-06-08:
Will geoclue provide the magnetic North heading?

hadess, 2015-06-08:
Ask Geoclue :)

volkov0aa, 2015-06-26:
I just realized that Geoclue provides the direction in which a device is moving, but not the direction in which a device is pointing.

hadess, 2015-06-26:
There's a geoclue mailing-list...

commit 53436ac9 (2015-06-03): main: Limit access to compass to Geoclue

Only geoclue can now access the Compass information, as we've moved it
to a separate object path and interface name.

Closes #29