hadess/iio-sensor-proxy #30: Dell 7352

sferris, 2015-06-21:
Hey There, I have a Dell 7352 and I believe it has the same chipset as the Yoga's:

$ lsusb | grep -i 0483:91D1
Bus 001 Device 002: ID 0483:91d1 STMicroelectronics 

And it seems to see it correctly:

$ G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy 
** (process:4208): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0001/HID-SENSOR-200041.7.auto/iio:device3
** (process:4208): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0001/HID-SENSOR-200041.7.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
** (process:4208): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.2.auto/iio:device4
** (process:4208): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.2.auto/iio:device4 of type accelerometer at IIO Buffer accelerometer
** (process:4208): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.4.auto/iio:device0
** (process:4208): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.4.auto/iio:device0 of type compass at IIO Buffer Compass
** (process:4208): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.2.auto/trigger4
** (process:4208): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0001/HID-SENSOR-200041.7.auto/trigger3
** (process:4208): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.4.auto/trigger0
** (process:4208): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4208): DEBUG: Got compass scale 1.000000

But it doesn't actually do anything?! Or, I'm missing something, which is more likely the case.. but I'm stuck. Any help?

Also: Fresh boot, and just compiled the latest kernel:

Linux lucy.virtualsmf.net 4.1.0-rc8 #1 SMP Sun Jun 21 09:59:25 MDT 2015 x86_64 x86_64 x86_64 GNU/Linux

Appreciate any help!

hadess, 2015-06-21:
What's "stuck"?

You need something driving iio-sensor-proxy. iio-sensor-proxy isn't an end-user tool. The API is available here:
https://developer.gnome.org/iio-sensor-proxy/1.0/

sferris, 2015-06-21:
Sorry hadess, I was just researching it further on my own and noticed that it's not used by gnome until 3.18.. my bad! (I'm on fedora 22) Thanks for the quick response though!

hadess, 2015-06-21:
That's right, there's no support for anything prior to GNOME 3.17...

hadess, 2015-06-21:
You can launch ```monitor-sensor``` to see outputs of all those sensors changing though.

closed 2015-06-21 by hadess