hadess/iio-sensor-proxy #31: Timestamping values

volkov0aa, 2015-06-23:
IIO provides timestamps for data. They could be useful for the accelerometer.
For example, Qt uses them for gesture recognizing.

hadess, 2015-07-10:
I couldn't figure out how to extract the timestamp from the readings unfortunately, patch would be welcome.

cheako, 2015-08-12:
It's because your going about this all wrong.  Firstly ditch uevent, that's for device additions/removals AFAIK, I'm not sure how you were able to use it effectively the way you have it but that all looks wrong.

The timestamps should come through as part of typical input event, you won't get them through EVIOCGABS as they don't apply to that ioctl.  EVIOCGABS is for use in first_values(void*), with a lock for if there's been a race and the data is now old.  If your code receives an event prior to handling the ioctl return, then my advice is to just ignore EVIOCGABS.  The d-bus API should mimic this behavior, a way to query the last value independent of the new events feed.  Given that you should save the last value, but don't push the return from EVIOCGABS as new/current.

Read https://www.kernel.org/doc/Documentation/input/input.txt specifically section 5.  As you'll likely need to handle dbus events then that'll restrict you to using nonblocking reads and select().  The axis data, complete with timestamps, will just be written to the device for your app to consume as regular file data.

hadess, 2015-10-09:
@cheako there's at least 2 other types of accelerometer drivers that don't use the input subsystem, there's another 4 drivers for light sensors and compasses that don't and can't use the input subsystem.

Also, we only use uevent to grab coarse orientation changes for accelerometers that use the pegatron accelerometer kernel API (eg. they send a kevent when the firmware tells them that the orientation changed).

We're going about it the right way. I'll refine my previous comment:
"I couldn't figure out how to extract the timestamp from the IIO ring buffer readings."

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/31

closed 2020-01-09 by hadess