hadess/iio-sensor-proxy #33: build failure on Debian sid

rickysarraf, 2015-06-24:
I'm trying to build this on Debian sid, and I get the following build failure.

rrs@learner:~/Community/iio-sensor-proxy (master)$ ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking whether make sets $(MAKE)... (cached) yes
checking for a sed that does not truncate output... /bin/sed
checking whether gcc understands -Wall... yes
checking whether gcc understands -Wstrict-prototypes... yes
checking whether gcc understands -Wnested-externs... yes
checking whether gcc understands -Werror=missing-prototypes... yes
checking whether gcc understands -Werror=implicit-function-declaration... yes
checking whether gcc understands -Werror=pointer-arith... yes
checking whether gcc understands -Werror=init-self... yes
checking whether gcc understands -Werror=format-security... no
checking whether gcc understands -Werror=format=2... yes
checking whether gcc understands -Werror=missing-include-dirs... yes
checking what warning flags to pass to the C compiler...  -Wall -Wstrict-prototypes -Wnested-externs -Werror=missing-prototypes -Werror=implicit-function-declaration -Werror=pointer-arith -Werror=init-self -Werror=format=2 -Werror=missing-include-dirs
checking what language compliance flags to pass to the C compiler... 
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for gtk-doc... yes
checking for gtkdoc-check... gtkdoc-check.test
checking for gtkdoc-check... /usr/bin/gtkdoc-check
checking for gtkdoc-rebase... /usr/bin/gtkdoc-rebase
checking for gtkdoc-mkpdf... /usr/bin/gtkdoc-mkpdf
checking whether to build gtk-doc documentation... no
checking for GTKDOC_DEPS... yes
checking for IIO_SENSOR_PROXY... yes
checking for gdbus-codegen... /usr/bin/gdbus-codegen
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating data/Makefile
config.status: creating src/Makefile
config.status: creating docs/Makefile
config.status: creating docs/version.xml
config.status: executing depfiles commands
16:06 ♒♒♒  ☺    
rrs@learner:~/Community/iio-sensor-proxy (master)$ make
Making all in data
make[1]: Entering directory '/media/SSHD/rrs-home/Community/iio-sensor-proxy/data'
  GEN      iio-sensor-proxy.service
make[1]: Leaving directory '/media/SSHD/rrs-home/Community/iio-sensor-proxy/data'
Making all in src
make[1]: Entering directory '/media/SSHD/rrs-home/Community/iio-sensor-proxy/src'
  GEN      iio-sensor-proxy-resources.c
  GEN      iio-sensor-proxy-resources.h
make  all-am
make[2]: Entering directory '/media/SSHD/rrs-home/Community/iio-sensor-proxy/src'
  CC       monitor_sensor-monitor-sensor.o
  CCLD     monitor-sensor
  CC       fake_input_accelerometer-fake-input-accelerometer.o
[... 108 lines cut ...]
iio_sensor_proxy-drv-iio-buffer-light.o: In function `driver_close':
/home/rrs/Community/iio-sensor-proxy/src/drivers.h:111: multiple definition of `driver_close'
iio_sensor_proxy-iio-sensor-proxy.o:/home/rrs/Community/iio-sensor-proxy/src/drivers.h:111: first defined here
iio_sensor_proxy-drv-iio-buffer-compass.o: In function `driver_discover':
/home/rrs/Community/iio-sensor-proxy/src/drivers.h:74: multiple definition of `driver_discover'
iio_sensor_proxy-iio-sensor-proxy.o:/home/rrs/Community/iio-sensor-proxy/src/drivers.h:74: first defined here
iio_sensor_proxy-drv-iio-buffer-compass.o: In function `driver_open':
/home/rrs/Community/iio-sensor-proxy/src/drivers.h:87: multiple definition of `driver_open'
iio_sensor_proxy-iio-sensor-proxy.o:/home/rrs/Community/iio-sensor-proxy/src/drivers.h:87: first defined here
iio_sensor_proxy-drv-iio-buffer-compass.o: In function `driver_set_polling':
/home/rrs/Community/iio-sensor-proxy/src/drivers.h:100: multiple definition of `driver_set_polling'
iio_sensor_proxy-iio-sensor-proxy.o:/home/rrs/Community/iio-sensor-proxy/src/drivers.h:100: first defined here
iio_sensor_proxy-drv-iio-buffer-compass.o: In function `driver_close':
/home/rrs/Community/iio-sensor-proxy/src/drivers.h:111: multiple definition of `driver_close'
iio_sensor_proxy-iio-sensor-proxy.o:/home/rrs/Community/iio-sensor-proxy/src/drivers.h:111: first defined here
collect2: error: ld returned 1 exit status
Makefile:459: recipe for target 'iio-sensor-proxy' failed
make[2]: *** [iio-sensor-proxy] Error 1
make[2]: Leaving directory '/media/SSHD/rrs-home/Community/iio-sensor-proxy/src'
Makefile:332: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/media/SSHD/rrs-home/Community/iio-sensor-proxy/src'
Makefile:343: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
16:06 ♒♒♒    ☹  => 2

rickysarraf, 2015-06-24:
I also checked out the 1.0 tag and the build failure is the same.

volkov0aa, 2015-06-24:
I guess inline functions should be declared as static.

rickysarraf, 2015-06-24:
Thank you @volkov0aa . That fixed the build.

@hadess I've tested it on my Lenovo Yoga 2 13 (Not Pro). Below is the output. You may want to add this model to your list.

rrs@learner:/tmp/iio$ monitor-sensor 
** Message: Accelerometer appeared
** Message: Accelerometer orientation changed: undefined
** Message: Light sensor appeared
** Message: Light changed: 0.000000 (lux)
^C

closed 2015-07-01 by hadess

rickysarraf, 2015-07-01:
BTW, I'll be packaging/maintaining it on Debian.

Here's the package description:

Package: iio-sensor-proxy
Architecture: linux-any
Depends: ${shlibs:Depends}, ${misc:Depends}, systemd 
Description: IIO sensors to D-Bus proxy
 Industrial I/O subsystem is intended to provide support for devices that
 in some sense are analog to digital or digital to analog convertors 
 .
 Devices that fall into this category are:
 * ADCs
 * Accelerometers
 * Gyros
 * IMUs
 * Capacitance to Digital Converters (CDCs)
 * Pressure Sensors
 * Color, Light and Proximity Sensors
 * Temperature Sensors
 * Magnetometers
 * DACs
 * DDS (Direct Digital Synthesis)
 * PLLs (Phase Locked Loops)
 * Variable/Programmable Gain Amplifiers (VGA, PGA)
 .
 Many ultrabooks ship with such devices. iio-sensor-proxy is intended to serve
 as a proxy, providing access to these devices through a D-Bus interface

hadess, 2015-07-01:
Except that's not what iio-sensor-proxy supports. It only supports ambient light, accelerometer and compasses.

commit 0974fe62 (2015-06-26): main: Fix build on Debian Sid

With help from @volkov0aa

Closes: #33