hadess/iio-sensor-proxy #34: HP Spectre x360

yalok, 2015-06-27:
Hi,

I am running ubuntu 15.04 on HP Spectre x360 with gnome 3.16. udevadm doesn't display anything like in readme example and it doesn't seem to work.
This is what I get in debug mode:

# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:2181): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:048D:8350.0002/HID-SENSOR-200073.0.auto/iio:device0
** (process:2181): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:048D:8350.0002/HID-SENSOR-200073.0.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:2181): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:048D:8350.0002/HID-SENSOR-200083.2.auto/iio:device2
** (process:2181): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:048D:8350.0002/HID-SENSOR-200083.2.auto/iio:device2 of type compass at IIO Buffer Compass
** (process:2181): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:048D:8350.0002/HID-SENSOR-200073.0.auto/trigger0
** (process:2181): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2181): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2181): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2181): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2181): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2181): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2181): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:048D:8350.0002/HID-SENSOR-200083.2.auto/trigger2
** (process:2181): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2181): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2181): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2181): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2181): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2181): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2181): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2181): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2181): DEBUG: Got compass scale 1.000000
** (process:2181): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2181): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2181): DEBUG: No new data available
** (process:2181): DEBUG: No new data available
** (process:2181): DEBUG: No new data available
** (process:2181): DEBUG: No new data available
** (process:2181): DEBUG: No new data available
.....
and so on

hadess, 2015-07-01:
You probably want to check whether that works better with the most recent kernel possible, say kernel 4.1.

hadess, 2015-07-25:
Please reopen if the problem still occurs with a more recent kernel (4.1 or 4.2rc kernels for example).

closed 2015-07-25 by hadess