hadess/iio-sensor-proxy #36: iio-sensor-proxy does nothing and just exits

aroig, 2015-07-05:
I'm testing `iio-sensor-proxy` on a Thinkpad Yoga. I get the following output, and then it just exits.

    $ sudo G_MESSAGES_DEBUG=all iio-sensor-proxy
    ** (process:25992): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:0483:91D1.0004/HID-SENSOR-200041.6.auto/iio:device5
    ** (process:25992): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:0483:91D1.0004/HID-SENSOR-200041.6.auto/iio:device5 of type ambient light sensor at IIO Buffer Light sensor
    ** (process:25992): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:0483:91D1.0004/HID-SENSOR-200073.1.auto/iio:device0
    ** (process:25992): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:0483:91D1.0004/HID-SENSOR-200073.1.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
    ** (process:25992): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:0483:91D1.0004/HID-SENSOR-200083.3.auto/iio:device2
    ** (process:25992): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:0483:91D1.0004/HID-SENSOR-200083.3.auto/iio:device2 of type compass at IIO Buffer Compass
    ** (process:25992): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:0483:91D1.0004/HID-SENSOR-200073.1.auto/trigger0
    ** (process:25992): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:0483:91D1.0004/HID-SENSOR-200041.6.auto/trigger5
    ** (process:25992): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:0483:91D1.0004/HID-SENSOR-200083.3.auto/trigger2
    ** (process:25992): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
    ** (process:25992): DEBUG: Got compass scale 1.000000
    ** (process:25992): DEBUG: iio-sensor-proxy is already running

It looks like the sensors are recognized, but `any_sensors_left (data)` returns `FALSE` in
`main()`. Any clue of what is going on?

Oh, by the way, I run this with kernel 4.0.7 on arch linux.

Thanks!

hadess, 2015-07-05:
> ```** (process:25992): DEBUG: iio-sensor-proxy is already running```
>  Any clue of what is going on?

Do you still need help, or you think you can figure it out? :)

aroig, 2015-07-05:
** (process:25992): DEBUG: iio-sensor-proxy is already running

It is definitely not running. I did read the logs I posted!

This message is printed from `name_lost_handler`, but not at the time `g_bus_own_name` is called, since the bulk of the debug messages I posted happen after dbus is setup.

From a quick look at the code, my guess was that `any_sensors_left (data))` returned FALSE for some reason, and dbus called the handler before releasing the bus name. I could have spent more time debugging this, but I preferred to ask first!

hadess, 2015-07-05:
> It is definitely not running. I did read the logs I posted! The systemd service triggered from udev also just exits.

Did you install the dbus service file in the correct location?

aroig, 2015-07-05:
Ah! that was it! I used `./configure --prefix=/usr`, and `make install` had put the file in `/usr/etc`.

Thanks a lot!

commit 92ccf349 (2015-07-05): main: Fix error message when we fail to own D-Bus service

As we can reach that handler if we're either running already, or if the
D-Bus .service file is not installed properly.

Closes #36
closed 2015-07-05 by hadess