hadess/iio-sensor-proxy #39: Fix accelerometer detection

hadess, 2015-07-26:
The BMA250 inside the WinBook TW100 isn't detected as an accelerometer because we expect the buffered/triggered sensor to be a HID sensor, and we hard-code names for the polling one. We need to do better.

cheako, 2015-10-14:
I found an issue where a string is freed from multiple locations while working on this.  Here are two patches, one that should correct the multiple frees and the other that adds strings for the INVN6500.

http://pastebin.com/SjUnemxu

cheako, 2015-10-17:
I've been stuck on the erratic data problem an haven't made any headway.

http://pastebin.com/k6GJi0uF

My first guess was a problem passing the correct bytes to be parsed, like an incorrect starting offset.  I'm fairly certain that is not an issue, I can now clearly see 24 bytes at a time being passed from the kernel when reading up to 4k.

http://pastebin.com/CGaY7MSS

My next trick will be to build a struct from the buffer_data.

cheako, 2015-10-17:
http://pastebin.com/qPaPQSgW

Logs of the buffer, it's all greek to me.

commit ccaf37e0 (2016-11-23): accel: Use udev tags to know which driver to use

This means that the buffer driver can now handle more than the
bog-standard HID sensor hub, and that the polling driver can handle any
IIO polling device.

Closes: #39
closed 2016-11-23 by hadess