hadess/iio-sensor-proxy #40: INVN6500 Sensor not detected

pinkflozd, 2015-08-01:
Hello. I'm tying to get your sensor proxy working on my Asus T100TAF, i'm running arch with systemd 223

P: /devices/platform/80860F41:02/i2c-2/i2c-INVN6500:00/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/platform/80860F41:02/i2c-2/i2c-INVN6500:00/iio:device0
E: DEVTYPE=iio_device
E: MAJOR=248
E: MINOR=0
E: SUBSYSTEM=iio
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=14379568

[pinkflozd@Bazdara-T100TAF ~]$ G_MESSAGES_DEBUG=all iio-sensor-proxy
** (process:1113): DEBUG: Could not find any supported sensors

How can i debug it further?
the sensor is getting data

[pinkflozd@Bazdara-T100TAF ~]$ cat /sys/bus/iio/devices/iio\:device0/in_accel_x_raw
-304
[pinkflozd@Bazdara-T100TAF ~]$ cat /sys/bus/iio/devices/iio\:device0/in_accel_y_raw
14304
[pinkflozd@Bazdara-T100TAF ~]$ cat /sys/bus/iio/devices/iio\:device0/in_accel_z_raw
-8092

hadess, 2015-08-03:
Duplicate of #39

closed 2015-08-03 by hadess

cheako, 2015-10-14:
This device is missing the in_anglvel_offset file.

http://lxr.free-electrons.com/source/drivers/iio/imu/inv_mpu6050/inv_mpu_core.c#L399

cheako, 2015-10-14:
https://launchpad.net/~cheako/+archive/ubuntu/t300chi/+sourcepub/5518385/+listing-archive-extra

There are two patches, one fixes a potential double g_free() and sets up for implementing multiple string matches.  The second adds the strings for this device so the device is enumerated.  The program quits after not being able to open in_anglvel_offset.

cheako, 2015-10-14:
root@debian:~# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy 
** (process:4391): DEBUG: Found accel_3d at /sys/devices/pci0000:00/INT3432:00/i2c-7/i2c-INVN6500:00/iio:device0
** (process:4391): DEBUG: Found device /sys/devices/pci0000:00/INT3432:00/i2c-7/i2c-INVN6500:00/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:4391): DEBUG: Found associated trigger at /sys/devices/pci0000:00/INT3432:00/i2c-7/i2c-INVN6500:00/trigger0

** (process:4391): WARNING **: Failed to read float from /sys/devices/pci0000:00/INT3432:00/i2c-7/i2c-INVN6500:00/iio:device0/in_anglvel_offset

** (process:4391): WARNING **: Problem reading scan element information: /sys/devices/pci0000:00/INT3432:00/i2c-7/i2c-INVN6500:00/iio:device0

cheako, 2015-10-14:
http://pastebin.com/k6GJi0uF#

cheako, 2015-10-15:
See [cheako](https://github.com/cheako)'s [mpu6050](https://github.com/cheako/iio-sensor-proxy/tree/mpu6050) branch