hadess/iio-sensor-proxy #41: warnings about dbus policy in Debian

rickysarraf, 2015-08-14:
@hadess 

On Debian, I see warnings and errors from lintian about the dbus policy that i-s-p has by default.
I haven't really spent much time investigating it from the dbus point of view yet, but am filing this bug to keep you informed. When/if I have more information, I'll update this bug report.

In case you think these errors and warnings are valid, then we can track it in this bug report.

W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy user="geoclue"><allow send_interface="net.hadess.SensorProxy.Compass" send_path="/net/hadess/SensorProxy/Compass"/>
N: 
N:    The package contains D-Bus policy configuration that uses one of the
N:    send_* conditions, but does not specify a send_destination, and is not
N:    specific to root.
N:    
N:    Rules of the form
N:    
N:      <allow send_interface="com.example.MyInterface"/>
N:    
N:    allow messages with the given interface to be sent to <any> service, not
N:    just the one installing the rule, which is rarely what was intended.
N:    
N:    Similarly, on the system bus, rules of the form
N:    
N:      <deny send_interface="com.example.MyInterface"/>
N:    
N:    are redundant with the system bus's default-deny policy, and have
N:    unintended effects on other services.
N:    
N:    This check ignores rules of the form
N:    
N:      <policy user="root">
N:        <allow ... />
N:      </policy>
N:    
N:    which are commonly used for the "agent" pattern seen in services like
N:    BlueZ and NetworkManager: a root-privileged daemon calls out to one or
N:    more per-user user interface agent processes with no specific name, so
N:    send_destination is not easily applicable. However, such rules should
N:    still be made as specific as possible to avoid undesired side-effects.
N:    
N:    Refer to https://bugs.freedesktop.org/show_bug.cgi?id=18961 and
N:    http://lists.freedesktop.org/archives/dbus/2008-February/009401.html for
N:    details.
N:    
N:    Severity: normal, Certainty: certain
N:    
N:    Check: dbus, Type: binary
N: 
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy user="geoclue"><allow send_interface="org.freedesktop.DBus.Introspectable" send_path="/net/hadess/SensorProxy/Compass"/>
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy user="geoclue"><allow send_interface="org.freedesktop.DBus.Properties" send_path="/net/hadess/SensorProxy/Compass"/>
E: iio-sensor-proxy: dbus-policy-excessively-broad etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy user="geoclue"><allow send_interface="org.freedesktop.DBus.Properties" send_path="/net/hadess/SensorProxy/Compass"/>
N: 
N:    The package contains D-Bus policy configuration that matches broad
N:    classes of messages. This will cause strange side-effects, is almost
N:    certainly unintended, and is a probable security flaw.
N:    
N:    For instance,
N:    
N:      <policy user="daemon">
N:        <allow send_type="method_call"/>
N:        <allow send_destination="com.example.Bees"/>
N:      </policy>
N:    
N:    in any system bus policy file would allow the daemon user to send any
N:    method call to any service, including method calls which are meant to be
N:    restricted to root-only for security, such as
N:    org.freedesktop.systemd1.Manager.StartTransientUnit. (In addition, it
N:    allows that user to send any message to the com.example.Bees service.)
N:    
N:    The intended policy for that particular example was probably more like
N:    
N:      <policy user="daemon">
N:        <allow send_type="method_call" send_destination="com.example.Bees"/>
N:      </policy>
N:    
N:    which correctly allows method calls to that particular service only.
N:    
N:    Refer to http://www.openwall.com/lists/oss-security/2015/01/27/25 for
N:    details.
N:    
N:    Severity: serious, Certainty: possible
N:    
N:    Check: dbus, Type: binary
N: 
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy user="geoclue"><allow send_interface="org.freedesktop.DBus.Peer" send_path="/net/hadess/SensorProxy/Compass"/>
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy context="default"><allow send_interface="net.hadess.SensorProxy"/>
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy context="default"><allow send_interface="org.freedesktop.DBus.Introspectable"/>
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy context="default"><allow send_interface="org.freedesktop.DBus.Properties"/>
E: iio-sensor-proxy: dbus-policy-excessively-broad etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy context="default"><allow send_interface="org.freedesktop.DBus.Properties"/>
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy context="default"><allow send_interface="org.freedesktop.DBus.Peer"/>
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy context="default"><deny send_interface="org.freedesktop.DBus.Introspectable" send_path="/net/hadess/SensorProxy/Compass"/>
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy context="default"><deny send_interface="org.freedesktop.DBus.Properties" send_path="/net/hadess/SensorProxy/Compass"/>
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy context="default"><deny send_interface="org.freedesktop.DBus.Peer" send_path="/net/hadess/SensorProxy/Compass"/>
W: iio-sensor-proxy: dbus-policy-without-send-destination etc/dbus-1/system.d/net.hadess.SensorProxy.conf <policy context="default"><deny send_interface="net.hadess.SensorProxy.Compass"/>
W: iio-sensor-proxy: binary-without-manpage usr/sbin/iio-sensor-proxy

smcv, 2017-02-02:
This is really bad, potentially a root security hole. The D-Bus policy installed by iio-sensor-proxy allows any connection to the system bus to send org.freedesktop.DBus.Properties.Set() method calls to any other connection to the system bus, including system services other than iio-sensor-proxy - even if those system services were only meant to be contactable by root.

If iio-sensor-proxy has the usual structure for a D-Bus service (one privileged process on the bus claims a bus name and provides services to other processes, similar to packagekit or polkit or udisks) then the correct solution is to add `send_destination="net.hadess.SensorProxy"` to every `<allow>` or `<deny>` rule, except for the one for `<allow own=.../>`. That restricts these rules so they only apply to messages sent to the connection that owns the name net.hadess.SensorProxy[1]. In particular, that prevents these rules from having an unintended effect on system services other than yours.

[1] Strictly speaking messages whose destination is any of the names of any connection that either owns that name, or is in the queue to own it when the current owner releases it

smcv, 2017-02-02:
Here is a non-weaponized exploit for this.

```
$ dbus-send --system --dest=org.freedesktop.Avahi --type=method_call --print-reply /org/freedesktop/Avahi org.freedesktop.DBus.Properties.Set string:Foo variant:string:bar
```

Failing test:

```
Error org.freedesktop.DBus.Error.UnknownObject: Method "Set" with signature "sv" on interface "org.freedesktop.DBus.Properties" doesn't exist
```

Passing test: you get a "permission denied" error instead

smcv, 2017-02-02:
Here is a non-weaponized exploit that is actually correct. Sorry about that, I picked a bad example.

You need NetworkManager, which I'm using because `nm_dispatcher` is a convenient example of something that only root is meant to be able to talk to.

Failing test: I can talk to nm_dispatcher as non-root, which I should not be able to do (but it's harmless in this case because nm_dispatcher has no properties)

```
$ dbus-send --system --dest=org.freedesktop.nm_dispatcher --type=method_call \
    --print-reply / org.freedesktop.DBus.Properties.Set string:Foo variant:string:bar                    
Error org.freedesktop.DBus.Error.UnknownMethod: No such interface
 'org.freedesktop.DBus.Properties' on object at path /
```

Passing test: dbus-daemon stops me

```
$ dbus-send --system --dest=org.freedesktop.nm_dispatcher --type=method_call \
    --print-reply / org.freedesktop.DBus.Properties.Set string:Foo variant:string:bar
Error org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched
 rules; type="method_call", sender=":1.5523" (uid=1000 pid=13527
 comm="dbus-send --system --dest=org.freedesktop.nm_dispa")
 interface="org.freedesktop.DBus.Properties" member="Set" error
 name="(unset)" requested_reply="0"
 destination="org.freedesktop.nm_dispatcher" (uid=0 pid=13528
 comm="/usr/lib/NetworkManager/nm-dispatcher ")
```

commit e2d81f21 (2017-02-02): security: Make D-Bus policy rules only affect SensorProxy itself

D-Bus policy XML files are generic configuration for the bus daemon:
they are conventionally named like a bus name, but there is nothing
that inherently limits their application to that bus name.

In particular this means that a rule like

  <policy context="default">
    <allow send_interface="org.freedesktop.DBus.Properties"/>

allows any process on the system bus to send an
org.freedesktop.DBus.Properties.Set() call to any other process on the
system bus, even if the destination process expected to be only
accessible by root. [... cut]
closed 2017-02-02 by hadess