hadess/iio-sensor-proxy #42: Not working on Onda v925W

drgnomage, 2015-08-23:
Hi there,

Can't seem to get it working! Onda v925W. Definately detected though!

** (process:1725): DEBUG: Found polling accelerometer at /sys/devices/platform/80860F41:04/i2c-4/i2c-SMO8500:00/iio:device0
** (process:1725): DEBUG: Found device /sys/devices/platform/80860F41:04/i2c-4/i2c-SMO8500:00/iio:device0 of type accelerometer at IIO Poll accelerometer
** (process:1725): DEBUG: iio-sensor-proxy is already running, or it cannot own its D-Bus name. Verify installation.

hadess, 2015-08-24:
You need to kill the existing instance of iio-sensor-proxy or, if it's not running already, install it in the correct location so that D-Bus daemon can find the dbus ".conf" file. It won't be able to own its name if the installation is in the wrong location.

drgnomage, 2015-08-24:
thank you for the reply, I followed the configuration instruction you provided. however I had to run autogen.sh to get configure, dont know if that would of changed what you had setup for the ./configure. also im using arch if thats relevant.

hadess, 2015-08-24:
The instructions are for building from the tarball, not from git.

Did you check whether iio-sensor-proxy was not already running?

drgnomage, 2015-08-24:
ah kk, yes it was running.

I had assumed rebooting would sort that out.

Now to get gnome 3.18 on arch.

hadess, 2015-08-24:
> ah kk, yes it was running.
>
> I had assumed rebooting would sort that out.

Well, if it was running on startup, it would still have been running.

> Your Usage section says gnome 3.18 or newer but the beta for 3.17 has just been released. Is this correct?

3.17 is just a beta for 3.18. You'll need a 3.17.x version indeed.

closed 2015-08-24 by hadess