hadess/iio-sensor-proxy #44: Dell XPS 12 (9Q23 Ivy bridge)

DorianVogel, 2015-09-14:
Hoy,
I'm trying to get this piece of SW for my laptop running 4.1.6 (ARCH)
udevadm lists various sensors properly producing something like that:
```
P: /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200083.5.auto/trigger2
E: DEVPATH=/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200083.5.auto/trigger2
E: SUBSYSTEM=iio
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=2697879

P: /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200086.4.auto
E: DEVPATH=/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200086.4.auto
E: DEVTYPE=mfd_device
E: DRIVER=hid_sensor_incl_3d
E: MODALIAS=platform:HID-SENSOR-200086
E: SUBSYSTEM=platform
```
However, when running monitor-sensor, I'm getting nothing.
Firing iioo-sensor-proxy by hand gives:
```
sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:16288): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200073.1.auto/iio:device4
** (process:16288): DEBUG: Found device /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200073.1.auto/iio:device4 of type accelerometer at IIO Buffer accelerometer
** (process:16288): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200083.5.auto/iio:device2
** (process:16288): DEBUG: Found device /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200083.5.auto/iio:device2 of type compass at IIO Buffer Compass
** (process:16288): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200073.1.auto/trigger4
** (process:16288): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:8087:0A05.0004/HID-SENSOR-200083.5.auto/trigger2
** (process:16288): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:16288): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
```
Then aside from this launching monitor-sensor gives me ```** (process:16288): DEBUG: No new data available``` in iio-sensor-proxy, and nothing in monitor-sensor.

Is this sufficient to observe sensor data, or is gnome mandatory ? I'm running plasma and not gnome.
Maybe a small hint on which file to use as OBJECT_PATH argument of dbus-send ?

(code has been built from git using the proposed flags for both ./autogen.sh and ./configure)

hadess, 2015-09-16:
You should use `monitor-sensor` to see the output from the device. In your case, it looks like the driver doesn't send out any information. Best try with a newer kernel. If it still fails to work with a newer kernel, you'll need to contact the linux-iio mailing-list.

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

closed 2016-11-22 by hadess