hadess/iio-sensor-proxy #45: no data on Lenovo Yoga 2 13

rickysarraf, 2015-10-04:
Now with GNOME 3.18, I hoped to see some action but unfortunate :-(
Turns out there might be issues with this make/model

```
rrs@learner:/usr/share/doc/iio-sensor-proxy$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:22187): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/0003:048D:8350.0003/HID-SENSOR-200041.9.auto/iio:device5
** (process:22187): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/0003:048D:8350.0003/HID-SENSOR-200041.9.auto/iio:device5 of type ambient light sensor at IIO Buffer Light sensor
** (process:22187): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/0003:048D:8350.0003/HID-SENSOR-200073.4.auto/iio:device0
** (process:22187): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/0003:048D:8350.0003/HID-SENSOR-200073.4.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:22187): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/0003:048D:8350.0003/HID-SENSOR-200083.6.auto/iio:device2
** (process:22187): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/0003:048D:8350.0003/HID-SENSOR-200083.6.auto/iio:device2 of type compass at IIO Buffer Compass
** (process:22187): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/0003:048D:8350.0003/HID-SENSOR-200073.4.auto/trigger0
** (process:22187): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:22187): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:22187): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:22187): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:22187): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:22187): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:22187): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/0003:048D:8350.0003/HID-SENSOR-200041.9.auto/trigger5
** (process:22187): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:22187): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:22187): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/0003:048D:8350.0003/HID-SENSOR-200083.6.auto/trigger2
** (process:22187): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:22187): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:22187): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:22187): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:22187): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:22187): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:22187): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:22187): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:22187): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:22187): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:22187): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:22187): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
** (process:22187): DEBUG: No new data available
[... 76 lines cut ...]
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};
2015-10-04 / 15:23:33 ♒♒♒  ☺    
```

Is it expected ? The Orientation is marked as 'undefined' ???
And monitor-sensor does not seem to be giving any output. I remember it did give meaningful output when I was initiailly packaging it.

rickysarraf, 2015-10-04:
Actually, dbus does see the event. But does not act on it, like changing the orientation of the desktop shell. I'm currently on GNOME Shell 3.17.92. Was this committed to 3.18 final ?

```
rrs@learner:/etc/dbus-1/system.d$ dbus-monitor 
signal time=1443956881.875758 sender=org.freedesktop.DBus -> destination=:1.62 serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameAcquired
   string ":1.62"
signal time=1443956881.875815 sender=org.freedesktop.DBus -> destination=:1.62 serial=4 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string ":1.62"
±signal time=1443956898.073538 sender=:1.4 -> destination=:1.28 serial=448 path=/org/gnome/Shell; interface=org.gnome.Shell; member=AcceleratorActivated
   uint32 107
   array [
      dict entry(
         string "device-id"
         variant             uint32 3
      )
      dict entry(
         string "timestamp"
         variant             uint32 514795
      )
      dict entry(
         string "action-mode"
         variant             uint32 1
      )
   ]
method call time=1443956898.074310 sender=:1.28 -> destination=:1.4 serial=280 path=/org/gnome/Shell; interface=org.gnome.Shell; member=ShowOSD
   array [
      dict entry(
         string "icon"
         variant             string "touchpad-disabled-symbolic"
      )
   ]
method return time=1443956898.079024 sender=:1.4 -> destination=:1.28 serial=449 reply_serial=280
signal time=1443956926.561803 sender=:1.4 -> destination=:1.28 serial=450 path=/org/gnome/Shell; interface=org.gnome.Shell; member=AcceleratorActivated
   uint32 106
   array [
      dict entry(
         string "device-id"
         variant             uint32 3
      )
      dict entry(
         string "timestamp"
         variant             uint32 543298
      )
      dict entry(
         string "action-mode"
         variant             uint32 1
      )
   ]
method call time=1443956926.562547 sender=:1.28 -> destination=:1.4 serial=281 path=/org/gnome/Shell; interface=org.gnome.Shell; member=ShowOSD
   array [
      dict entry(
         string "icon"
         variant             string "input-touchpad-symbolic"
      )
   ]
method return time=1443956926.568494 sender=:1.4 -> destination=:1.28 serial=451 reply_serial=281

```

hadess, 2015-10-04:
```AcceleratorActivated``` isn't an accelerometer, it's a key combination. Somebody pressed the touchpad disable button on the keyboard.

The problem you're having is that the kernel doesn't send any new information. Try the latest one.

rickysarraf, 2015-10-04:
Hmmm. I am on 4.2.3 already. I'll try the latest 4.3 rc

s3nt fr0m a $martph0ne, excuse typ0s

rickysarraf, 2015-10-04:
Sigh!! All these month I was under the impression that the kernel did have the necessary driver for this device. But looks like that is not the case.

hadess, 2015-10-05:
So, this is fixed?

rickysarraf, 2015-10-05:
@hadess THanks buddy. I had to patch the kernel to add the quirk, after which, just a charm. Thanks for your integration of iio-sensor-proxy with GNOME. :-)

Just for the sake of other users, if they run into similar issues.

```
From 78466e63250f12f4ed006dff75c23248320e4cf0 Mon Sep 17 00:00:00 2001
From: Ritesh Raj Sarraf <rrs@debian.org>
Date: Sun, 4 Oct 2015 22:39:35 +0530
Subject: [PATCH] Add quirk for Lenovo Yoga 2 with ITE Chips

---
 drivers/hid/hid-ids.h        | 1 +
 drivers/hid/hid-sensor-hub.c | 3 +++
 2 files changed, 4 insertions(+)

diff --git a/drivers/hid/hid-ids.h b/drivers/hid/hid-ids.h
index f769208..127c777 100644
--- a/drivers/hid/hid-ids.h
+++ b/drivers/hid/hid-ids.h
@@ -510,6 +510,7 @@
 
 #define USB_VENDOR_ID_ITE               0x048d
 #define USB_DEVICE_ID_ITE_LENOVO_YOGA   0x8386
+#define USB_DEVICE_ID_ITE_LENOVO_YOGA2  0x8350
 
 #define USB_VENDOR_ID_JABRA		0x0b0e
 #define USB_DEVICE_ID_JABRA_SPEAK_410	0x0412
diff --git a/drivers/hid/hid-sensor-hub.c b/drivers/hid/hid-sensor-hub.c
index a76eb2a..05afa25 100644
--- a/drivers/hid/hid-sensor-hub.c
+++ b/drivers/hid/hid-sensor-hub.c
@@ -777,6 +777,9 @@ static const struct hid_device_id sensor_hub_devices[] = {
 	{ HID_DEVICE(HID_BUS_ANY, HID_GROUP_SENSOR_HUB, USB_VENDOR_ID_ITE,
 			USB_DEVICE_ID_ITE_LENOVO_YOGA),
 			.driver_data = HID_SENSOR_HUB_ENUM_QUIRK},
+	{ HID_DEVICE(HID_BUS_ANY, HID_GROUP_SENSOR_HUB, USB_VENDOR_ID_ITE,
+			USB_DEVICE_ID_ITE_LENOVO_YOGA2),
+			.driver_data = HID_SENSOR_HUB_ENUM_QUIRK},
 	{ HID_DEVICE(HID_BUS_ANY, HID_GROUP_SENSOR_HUB, HID_ANY_ID,
 		     HID_ANY_ID) },
 	{ }
-- 
2.5.3
```

closed 2015-10-05 by rickysarraf

hadess, 2015-10-05:
Did you send the patch upstream?

rickysarraf, 2015-10-05:
On Mon, 2015-10-05 at 04:24 -0700, hadess wrote:
> Did you send the patch upstream?

Yup!! Jiri Kosina is the maintainer and has already asked the reason to
add the quirk. Hopefully, he'll include it.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."