hadess/iio-sensor-proxy #46: Dell XPS 13 (Ivy Bridge)

lviggiani, 2015-10-09:
I'm on arch linux with kernel 4.2.2
I've installed iio-sensor-proxy for AUR.
I have:

    $udevadm info --export-db | grep iio:device0
    P:/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:01/PNP0C09:00/ACPI0008:00/iio:device0
    N: iio:device0
    E: DEVNAME=/dev/iio:device0
    E: DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:01/PNP0C09:00/ACPI0008:00/iio:device0

and

    $ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
    Errore: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files

Service

    $ systemctl status iio-sensor-proxy.service
    ● iio-sensor-proxy.service - IIO Sensor Proxy service
       Loaded: loaded (/usr/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: disabled)
       Active: inactive (dead) since ven 2015-10-09 14:08:37 CEST; 14min ago
       Process: 420 ExecStart=/usr/bin/iio-sensor-proxy (code=exited, status=0/SUCCESS)
       Main PID: 420 (code=exited, status=0/SUCCESS)
    
    ott 09 14:08:37 dell-xps13-luca systemd[1]: Starting IIO Sensor Proxy service...
    ott 09 14:08:37 dell-xps13-luca systemd[1]: Started IIO Sensor Proxy service.
    Hint: Some lines were ellipsized, use -l to show in full.

Finally debug

    G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
    ** (process:4207): DEBUG: Could not find any supported sensors

However

    $ cat /sys/bus/acpi/devices/ACPI0008\:00/iio\:device0/in_illuminance_raw 
   113

outputs current illuminance

hadess, 2015-10-09:
Is there a file called ```in_illuminance_input``` in that directory? What's the content of the ```name``` file in that directory? Can you attach the output of ```ls -R``` in that directory as well, to see if it has triggers?

lviggiani, 2015-10-09:
no `in_illuminance_input` :(

````
$ ls -R /sys/bus/acpi/devices/ACPI0008\:00/iio\:device0/
/sys/bus/acpi/devices/ACPI0008:00/iio:device0/:
buffer  dev  in_illuminance_raw  name  power  scan_elements  subsystem  uevent

/sys/bus/acpi/devices/ACPI0008:00/iio:device0/buffer:
enable  length  watermark

/sys/bus/acpi/devices/ACPI0008:00/iio:device0/power:
async                 runtime_active_kids  runtime_status
autosuspend_delay_ms  runtime_active_time  runtime_suspended_time
control               runtime_enabled      runtime_usage

/sys/bus/acpi/devices/ACPI0008:00/iio:device0/scan_elements:
in_illuminance_en  in_illuminance_index  in_illuminance_type
````
and
````
$ cat /sys/bus/acpi/devices/ACPI0008\:00/iio\:device0/name
acpi-als
````

lviggiani, 2015-10-13:
any feedback? is that a kernel bug?

elad661, 2015-10-20:
It's not necessarily a kernel bug, as this could be a buffering sensor and not a polling sensor. I have a hunch this would work with only some minor changes to `drv-iio-buffer-light.c`. The two main differences is that the scan element is named `in_illuminance` and not `in_intensity_both`, and that the `name` field is `acpi-als` instead of just `als`.

Since I don't have the hardware to test with (I have the newer XPS 13, but it doesn't seem to have a light sensor unfortunately), I can't verify this myself, and this is already a long shot because there might be other behavioural differences that I don't know about.

@lviggiani, can you please try to edit `drv-iio-buffer-light.c` replace `in_intensity_both` in line 54 with `in_illuminance` and replace `als` with `acpi-als` in line 157? after you do this, compile and install, then run the main binary with debugging enabled again. I'm not sure this will work, it will probably fail in some weird way, but that's the first step of identifying the fix for this issue.

lviggiani, 2015-10-21:
Something changed. After patching `rv-iio-buffer-light.c`, recompiling and running the binary in debug mode, this is what I get:

````
$ G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:18838): DEBUG: Found als at /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:01/PNP0C09:00/ACPI0008:00/iio:device0
** (process:18838): DEBUG: Found device /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:01/PNP0C09:00/ACPI0008:00/iio:device0 of type ambient light sensor at IIO Buffer Light sensor

** (process:18838): WARNING **: Could not find trigger name associated with /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:01/PNP0C09:00/ACPI0008:00/iio:device0
````

elad661, 2015-10-21:
Yeah, it makes sense monitor-sensor won't show anything, it won't show anything until we get this to work. 

I have another guess that you can try, can you try changing `als-dev` to `acpi-als-dev` in line 118 of the same file? (and keep the other changes you made, of course). 

I don't know if this is the correct trigger name (and maybe my guess that it's a buffering sensor was wrong), but it's worth a shot. That's the best I can do to help without having actual hardware to test on.

lviggiani, 2015-10-21:
...still the same... :(

elad661, 2015-10-21:
Reading the kernel driver for acpi-als shows it's indeed not a triggering sensor (unlike hid-als). I don't know if it can be changed to be triggering. drv-iio-buffer-light.c won't work with this kind of sensor.

I wonder how you're supposed to read from this sensor. I guess the first step is probably enabling the buffer using the `enable` file you have there? perhaps you can write your own driver for iio-sensor-proxy to support this kind of sensor... I also wonder if the kernel driver can be changed to work more similarly to `hid-als` or to one of the polling drivers. Might be worth to ask the linux-iio mailing list.

dberg918, 2015-10-23:
Not sure if this helps, but I have an ASUS Zenbook UX305 that also uses the acpi-als kernel driver. Someone wrote a controller for the light sensor in a similar laptop (Zenbook UX32VD), which is on [GitHub](https://github.com/danieleds/Asus-Zenbook-Ambient-Light-Sensor-Controller). Might give some clues on how to enable, poll, and manipulate a sensor like this.

There's also a separate kernel driver just for the Zenbook [here](https://github.com/victorenator/als), though I'm not sure how useful it is given that the kernel seems to detect the sensor and use the acpi-als driver by default.

dberg918, 2015-10-23:
As for reading from the sensor, `in_illuminance_raw` in `/sys/bus/acpi/devices/ACPI0008:00/iio:device0/` is the file that reports ambient light.

elad661, 2015-10-23:
@dberg918 the app you linked to uses an out-of-tree kernel driver for acpi-als, which exposes the sensor in a way very different than iio. When the acpi-als was submitted upstream, it was re-written to have the same API  as any other iio sensor. Therefor, this app is not helpful.

Regarding reading from raw, this is probably not a good idea. Why don't do this for any other sensor type.

dberg918, 2016-02-10:
Just FYI, the staging commit for [kernel 4.5-rc3][rc3] contains a fix for the acpi-als driver (same as the reference above). The data coming from the sensor was being reported as raw, when it was in fact already processed to lux.

I tested it on my ASUS UX305 and it works; the sensor is detected, the Automatic Brightness switch in GNOME 3.18 appears, and turning it on changes the backlight automatically. The changes in brightness are a bit choppy and distracting, and the screen dims a bit too much with indoor overhead lighting, but I'm not sure these are iio-sensor-proxy issues.

[rc3]: https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=dacd53c80525809cc9ce85e3a4ad409d15209f2a
   "Merge tag 'staging-4.5-rc3'"

hadess, 2016-04-19:
I'm leaving this opened until I hear back from Gabriele so he can tell me whether there's something we should have changed in iio-sensor-proxy.

hadess, 2016-04-19:
It was a bug in the driver indeed. Thanks to Gabriele for fixing it!

closed 2016-04-19 by hadess