hadess/iio-sensor-proxy #51: Yoga 2 Pro gnome-settings-daemon stops working after a while

vladimiroltean, 2015-10-19:
Hi, I didn't know of a better place to post this.
I am using Arch Linux with GNOME 3.18, and this proxy to provide screen auto rotation.
After a while however (can't pinpoint what's causing it), auto rotation stops working. I have checked the output of monitor-sensor and it is still running, giving messages such as:
> ** Message: Accelerometer orientation changed: left-up

Either reloading GNOME shell (Alt-F2 + r) or running "killall gnome-settings-daemon" help (temporarily) solve the issue, but I'd much rather prefer it not happening.
The output of "G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy" is:
> ** (process:7135): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/iio:device5
> ** (process:7135): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/iio:device5 of type ambient light sensor at IIO Buffer Light sensor
> ** (process:7135): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/iio:device3
> ** (process:7135): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
> ** (process:7135): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/iio:device2
> ** (process:7135): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/iio:device2 of type compass at IIO Buffer Compass
> ** (process:7135): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/trigger3
> ** (process:7135): WARNING **: Failed to write current_trigger file Permission denied
> ** (process:7135): WARNING **: Could not open for write '/sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/iio:device3/buffer/enable'
> ** (process:7135): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/trigger5
> ** (process:7135): WARNING **: Failed to write current_trigger file Permission denied
> ** (process:7135): WARNING **: Could not open for write '/sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/iio:device5/buffer/enable'
> ** (process:7135): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/trigger2
> ** (process:7135): WARNING **: Failed to write current_trigger file Permission denied
> ** (process:7135): WARNING **: Could not open for write '/sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/iio:device2/buffer/enable'

hadess, 2015-10-19:
```
** (process:7135): WARNING *: Failed to write current_trigger file Permission denied
* (process:7135): WARNING *: Could not open for write '/sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/iio:device3/buffer/enable'
```

This is a problem. Is the kernel producing warnings? As which user are you running this helper?

vladimiroltean, 2015-10-19:
Directly as a login user.

hadess, 2015-10-19:
That won't work, the iio-sensor-proxy needs to run as root.

vladimiroltean, 2015-10-19:
Yes, sorry for the false alarm, running as root produces apparently correct results in the log:
> ** (process:12301): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/iio:device0
> ** (process:12301): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/iio:device0 of type ambient light sensor at IIO Buffer Light sensor
> ** (process:12301): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/iio:device3
> ** (process:12301): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
> ** (process:12301): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/iio:device4
> ** (process:12301): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/iio:device4 of type compass at IIO Buffer Compass
> ** (process:12301): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.2.auto/trigger3
> ** (process:12301): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.9.auto/trigger0
> ** (process:12301): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
> ** (process:12301): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
> ** (process:12301): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.4.auto/trigger4
> ** (process:12301): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
> ** (process:12301): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
> ** (process:12301): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
> ** (process:12301): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
> ** (process:12301): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
> ** (process:12301): DEBUG: No new data available
> ** (process:12301): DEBUG: Read from IIO: 31, -1008, 45
> ** (process:12301): DEBUG: Accel sent by driver (quirk applied): 0, 9, 0
> ** (process:12301): DEBUG: Emitted orientation changed: from undefined to normal
> ** (process:12301): DEBUG: No new data available
> ** (process:12301): DEBUG: Read from IIO: 32, -1009, 46
> ** (process:12301): DEBUG: Accel sent by driver (quirk applied): 0, 9, 0
> ** (process:12301): DEBUG: No new data available
> ** (process:12301): DEBUG: Read from IIO: 31, -1009, 48
> ** (process:12301): DEBUG: Accel sent by driver (quirk applied): 0, 9, 0
> ** (process:12301): DEBUG: No new data available
> ** (process:12301): DEBUG: Read from IIO: 41, -1009, 45
> ** (process:12301): DEBUG: Accel sent by driver (quirk applied): 0, 9, 0
> ** (process:12301): DEBUG: No new data available
> ** (process:12301): DEBUG: Read from IIO: 47, -1009, 45
> ** (process:12301): DEBUG: Accel sent by driver (quirk applied): 0, 9, 0
> ** (process:12301): DEBUG: No new data available
> ** (process:12301): DEBUG: Read from IIO: 68, -1008, 53
> ** (process:12301): DEBUG: Accel sent by driver (quirk applied): 0, 9, 0
> ** (process:12301): DEBUG: Read from IIO: 78.100001
[... cut ...]
> [   30.231433]  [<ffffffffa049e420>] ? drm_mode_setplane+0x1b0/0x1b0 [drm]
> [   30.231437]  [<ffffffff811e29e5>] do_vfs_ioctl+0x295/0x480
> [   30.231440]  [<ffffffff811ecad7>] ? __fget+0x77/0xb0
> [   30.231443]  [<ffffffff811e2c49>] SyS_ioctl+0x79/0x90
> [   30.231446]  [<ffffffff8108300e>] ? SyS_rt_sigprocmask+0x8e/0xc0
> [   30.231449]  [<ffffffff8157162e>] entry_SYSCALL_64_fastpath+0x12/0x71
> [   30.231451] ---[ end trace 9f06c7cf1835b7b7 ]---

As can be seen, I am running the latest stable kernel, 4.2.3. I will also try the mainline version.

hadess, 2015-10-19:
```/usr/lib/gnome-settings/daemon/gsd-test-orientation``` will only run if you disabled the orientation plugin. Re-enable it and it should pick up the changes. Make sure to use GNOME 3.18, it won't work with older versions.

vladimiroltean, 2015-10-28:
After more investigation, it seems that this piece of software works correctly, with all debug messages showing what they should. However, auto rotation in GNOME doesn't properly work at startup (after 5-6 successful rotations it stops), until I restart the shell (Alt-F2 + r). It works completely fine after that (including after waking up from sleep).
I am trying to run the following immediately after reboot:
> gsettings set org.gnome.settings-daemon.plugins.orientation active false
> /usr/lib/gnome-settings/daemon/gsd-test-orientation

It shows the following sample output:
> (gsd-test-orientation:29957): orientation-plugin-DEBUG: Found orientation 'normal' for accelerometer
> (gsd-test-orientation:29957): orientation-plugin-DEBUG: Orientation changed to 'normal', switching screen rotation
> (gsd-test-orientation:29957): orientation-plugin-DEBUG: Found orientation 'left-up' for accelerometer
(gsd-test-orientation:29957): orientation-plugin-DEBUG: Orientation changed to 'left-up', switching screen rotation
> (gsd-test-orientation:29957): orientation-plugin-DEBUG: Found orientation 'normal' for accelerometer
> (gsd-test-orientation:29957): orientation-plugin-DEBUG: Orientation changed to 'normal', switching screen rotation

Unfortunately, after running "gsettings set org.gnome.settings-daemon.plugins.orientation active true" again, the effective screen rotation does not happen.
I am moving to the gnome-settings-daemon issues page, since I believe the bug to be there.

hadess, 2016-04-19:
As iio-sensor-proxy is reading the sensor correctly, reporting it to the desktop to use, and that gsd-test-orientation can read the data, it looks like a GNOME bug, not one in iio-sensor-proxy. Please file a bug against gnome-settings-daemon at https://bugzilla.gnome.org referencing this bug, and make sure that you're using the newest stable version of gnome-settings-daemon as well.

closed 2016-04-19 by hadess