hadess/iio-sensor-proxy #54: Running the dbus service as root is (probably?) a security risk

badshah400, 2015-11-06:
Hi!
It seems that running the dbus service as root is a security risk and in conflict with openSUSE dbus service policy. Please take a look at
https://bugzilla.opensuse.org/show_bug.cgi?id=939191
and let us know if there is anything that can be done to work around this issue or if there really is no option other than running the dbus service as root.

According to the security team's comment on the issue:

>The question is whether its really needed to run as root.
It looks like the service just reads some /sys files and turns it
over via dbus. Can you elaborate wther this really requires root perms?
(e.g. what the permissions of these particular files are; I dont have
a tablet to check for).
> So it might be sufficient to read them out as user or to be in the
uid 0 group.
>
> We dont want to have yet another dbus service running as root.

That said, thanks a lot for the software. It works great on my computer (which is why I am trying to get it accepted into the official openSUSE repositories).

Package version: 1.1

hadess, 2015-11-06:
Your security team doesn't know how to read code, seriously. It needs to be able to read *and write* to sysfs files because the IIO user-space API absolutely sucks.

I wouldn't have minded commenting on this if I felt there was even a modicum of work done, rather than brushing over the code in express time.

closed 2015-11-06 by hadess

badshah400, 2015-11-06:
Thanks for your prompt response.