hadess/iio-sensor-proxy #55: Updating to kernel 4.3.0 made iio-sensor-proxy stop working

badshah400, 2015-11-10:
I recently updated to kernel 4.3.0. Previously with kernel 4.2.4, iio-sensor-proxy was working great: auto-rotation, auto-brightness, the whole works. However immediately after booting into kernel 4.3.0, everything stopped working. Output of G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy running as root (and below that the output from same command but using Kernel 4.2.4), done after stopping the iio-sensor-proxy service as suggested in the README.md is pasted below. Please let me know if this is a kernel issue and does not belong here.

**Kernel 4.3.0**

> ** (process:2088): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.8.auto/iio:device0
** (process:2088): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.8.auto/iio:device0 of type ambient light sensor at IIO Buffer Light sensor
** (process:2088): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.1.auto/iio:device1
** (process:2088): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.1.auto/iio:device1 of type accelerometer at IIO Buffer accelerometer
** (process:2088): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.3.auto/iio:device5
** (process:2088): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.3.auto/iio:device5 of type compass at IIO Buffer Compass
** (process:2088): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.1.auto/trigger1
** (process:2088): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.8.auto/trigger0
** (process:2088): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2088): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2088): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.3.auto/trigger5
** (process:2088): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2088): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2088): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available
** (process:2088): DEBUG: No new data available

**Kernel 4.2.4**

>** (process:17505): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.8.auto/iio:device2
** (process:17505): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.8.auto/iio:device2 of type ambient light sensor at IIO Buffer Light sensor
** (process:17505): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.1.auto/iio:device0
** (process:17505): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.1.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:17505): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.3.auto/iio:device1
** (process:17505): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.3.auto/iio:device1 of type compass at IIO Buffer Compass
** (process:17505): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200073.1.auto/trigger0
** (process:17505): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17505): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17505): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17505): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17505): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17505): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:17505): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200041.8.auto/trigger2
** (process:17505): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:17505): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:17505): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2047:0855.0001/HID-SENSOR-200083.3.auto/trigger1
** (process:17505): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
[... 19 lines cut ...]
** (process:17505): DEBUG: No new data available
** (process:17505): DEBUG: Read from IIO: 4, -964, -318
** (process:17505): DEBUG: Accel sent by driver (quirk applied): 0, 9, -3
** (process:17505): DEBUG: No new data available
** (process:17505): DEBUG: Read from IIO: 4, -963, -314
** (process:17505): DEBUG: Accel sent by driver (quirk applied): 0, 9, -3
** (process:17505): DEBUG: No new data available
** (process:17505): DEBUG: No new data available
** (process:17505): DEBUG: No new data available
[... cut ...]
** (process:17505): DEBUG: No new data available
** (process:17505): DEBUG: Read from IIO: 4, -961, -311
** (process:17505): DEBUG: Accel sent by driver (quirk applied): 0, 9, -3
** (process:17505): DEBUG: No new data available
** (process:17505): DEBUG: Read from IIO: 8, -963, -333
** (process:17505): DEBUG: Accel sent by driver (quirk applied): 0, 9, -3
** (process:17505): DEBUG: No new data available
** (process:17505): DEBUG: Read from IIO: 3, -963, -310
** (process:17505): DEBUG: Accel sent by driver (quirk applied): 0, 9, -3

hadess, 2015-11-10:
If using a newer kernel is the only difference, that strikes me as a kernel bug I'm afraid.

badshah400, 2015-11-11:
Thanks @hadess, I just opened a bug against the 4.3.0 kernel over at openSUSE's bugzilla:
https://bugzilla.opensuse.org/show_bug.cgi?id=954588

closed 2016-01-11 by hadess

redbmk, 2016-01-11:
@badshah400 I wonder if this is related to #66 and #62. Does it work after you wake from suspend?

badshah400, 2016-01-11:
Hey, thanks! This is actually true (I was still using kernel 4.2.x where it works right out of boot) but on testing just now with kernel 4.3.3, everything does work after waking from suspend.

redbmk, 2016-01-11:
Good to know! I'll keep an eye on the bug you filed and I guess just downgrade my kernel for now.