hadess/iio-sensor-proxy #57: Cannot calibrate accelerator data to the right orientation

Icenowy, 2015-11-12:
I'm using iio-sensor-proxy-1.1 and gnome-3.18 on Teclast X98 Air 3G.
The accelorator is recognized, and it surely works.
However, the orientation is always 90 degress anticlockwise to the right orientation indicated by the position of the tablet.
I didn't find any conf files to fix it.
(Under Windows, this issue is fixed by modifying the registry)

sassmann1, 2015-11-21:
Same issue with Baytrail-m tablet on Fedora 23.
AXDIA International GmbH Wintab Gen 8/Type2

The screen rotates properly but the orientation is always 90° off. Any idea how to fix this?

francescor, 2016-01-13:
Similar problem with Asus trasformer book Flip TP550LJ with Fedora 23.  
Thanks to today's fedora 23 kernel upgrade the accelerometer started to work, but screen is inverted (upside-down): had to type

```bash
xrandr -o normal
```
or to stop  iio-sensor-proxy.service
```bash
systemctl start iio-sensor-proxy.service
```

any ideas on how to correct the issue?
Thanks for your code!

hadess, 2016-04-19:
@Icenowy this is a bug in the kernel driver:
https://bugs.freedesktop.org/show_bug.cgi?id=94894

@sassmann1 if the graphics change orientation when X (or plymouth if your distribution uses that) starts, then it's the above bug, otherwise, please file a separate bug.

@francescor same thing, please file a new bug if the orientation of the graphics change on the screen during boot up.

closed 2016-04-19 by hadess

Icenowy, 2016-06-17:
@hadess I found that your bug is not right...
My issue is 90 degree rotation, but the kernel bug is upside down.

Icenowy, 2016-06-17:
my tablet's i915 fb is also upside down, but the orientation of iio-sensor-proxy is 90 degree off.

hadess, 2016-11-23:
> my tablet's i915 fb is also upside down, but the orientation of iio-sensor-proxy is 90 degree off.

Once the framebuffer is the right way around, we can apply a quirk for the accelerometer's orientation. We're currently working on that, but we won't merge it before the framebuffer orientation is fixed, otherwise we'll need to change the quirk again afterwards.

dayara, 2017-04-23:
Same problem on Cube Mix Plus. Portrait reversed.

hadess, 2017-04-24:
This bug is closed.

Icenowy, 2017-04-24:
Thus is the quirk said added?

If not, you should re-open it before it's really resolved.

Your closing is based on wrong assumption, so it's wrong.

hadess, 2017-04-24:
> Thus is the quirk said added?

No

> If not, you should re-open it before it's really resolved.

No, because the original bug in the kernel wasn't fixed.

dayara, 2017-04-24:
But how do I fix the bug?

francescor, 2017-04-24:
for me, too
now with fedora 25 it rotates 90 degree clockwise as stated by@Icenowy  
(Asus trasformer book Flip TP550LJ, which is NOT a laptop I would recommend, btw)

dayara, 2017-04-28:
This script works for me:

#!/bin/sh
# Auto rotate screen based on device orientation

# Receives input from monitor-sensor (part of iio-sensor-proxy package)
# Screen orientation and launcher location is set based upon accelerometer position
# Launcher will be on the left in a landscape orientation and on the bottom in a portrait orientation
# This script should be added to startup applications for the user

# Clear sensor.log so it doesn't get too long over time
> sensor.log

# Launch monitor-sensor and store the output in a variable that can be parsed by the rest of the script
monitor-sensor >> sensor.log 2>&1 &

# Parse output or monitor sensor to get the new orientation whenever the log file is updated
# Possibles are: normal, bottom-up, right-up, left-up
# Light data will be ignored
while inotifywait -e modify sensor.log; do
# Read the last line that was added to the file and get the orientation
ORIENTATION=$(tail -n 1 sensor.log | grep 'orientation' | grep -oE '[^ ]+$')

# Set the actions to be taken for each possible orientation
case "$ORIENTATION" in
normal)
xrandr --output eDP1 --rotate normal;; #instead of --rotate normal
bottom-up)
xrandr --output eDP1 --rotate inverted;; #instead of --rotate inverted
right-up)
xrandr --output eDP1 --rotate left;; #instead of --rotate right
left-up)
xrandr --output eDP1 --rotate right;; #instead of --rotate left
esac
done

dayara, 2017-04-28:
Someone has deleted my previous post with a script that fixes iio-sensor-proxy errors on my Cube Mix Plus.
This script rotates the screen and the coordinates of the stylus correctly.

[rotation.zip](https://github.com/hadess/iio-sensor-proxy/files/965536/rotation.zip)