hadess/iio-sensor-proxy #58: Surface Pro 3: accel/light sensors detected but not working (Fedora 23)

smithfred, 2015-11-12:
On a Surface Pro 3 running Fedora 23 (running F23 kernels with minor patches to enable other hardware (which shouldn't affect sensors): https://github.com/shvr/fedora-surface-pro-3-kernel), accelerometer and light sensors seem to be detected but don't seem to work.

monitor-sensors doesn't output anything when orientation/ambient light are changed.

gdus introspect output:

```
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
```

Please let me know what info would be useful to debug this. Thanks!

hadess, 2015-11-17:
> Please let me know what info would be useful to debug this. Thanks!

How about the debug info in the README?

smithfred, 2015-11-17:
Thanks for the reply on here. I will do that shortly. In the meantime, I have just found that cat-ing the raw accel. data from /sys/bus/iio/devices/iio:device[n]/in_accel_x_raw (or y or z) causes iio-sensor-proxy to pick up correct accelerometer data. After a matter of a minute or two, iio-sensor-proxy then lapses into failing again. Seemingly if I do watch cat on one of the files, it keeps it working.

Additionally, the iio device number is sometimes different after a reboot, e.g. 3 vs. 4.

smithfred, 2015-11-17:
[debug.txt](https://github.com/hadess/iio-sensor-proxy/files/36664/debug.txt)

"No new data available" is repeated every half second.

Line 40/41 were output when I cat-ed in_accel_x_raw once (see previous comment).

smithfred, 2015-11-17:
The same behaviour occurs with the light sensor - "No new data available" is shown regardless of ambient light changes (i.e. covering the sensor) until I cat the relevant file - in this case in_intensity_both_raw, at which point I get a "Light level sent by driver" line and a "Emitted light changed", the value of which seems to map in a sensible way to the actual light level.

smithfred, 2015-12-06:
Is there any further information needed at this point to diagnose this issue?

schuhumi, 2015-12-13:
The exact same problem appears for me on an HP Envy X2 13-j001ng running Arch Linux (Kernel 4.2.5-1). Nothing happens until cat-ing an in_accel_{x,y,z}_raw file, then it works one single time just like intended. Every update has to be triggered by cat-ing the file.

[debug.txt](https://github.com/hadess/iio-sensor-proxy/files/60614/debug.txt)

schuhumi, 2015-12-14:
As far as I can see iio-sensor-proxy /dev/iio:device0 (0 happened to be the one for me right now, /dev is the important bit), thus it expects continuous mode (see https://community.freescale.com/message/401375#401375 ). It doesn't get any data, which I verified using cat. I'm not yet sure what sensor exactle my device has / what driver is being used, but I had a look at this to get an idea: http://processors.wiki.ti.com/index.php/AM335x_ADC_Driver's_Guide

All continous mode flags I could find are set to reasonable values:
 
    [simon@hptablet ~]$ cat /sys/bus/iio/devices/iio\:device0/in_accel_sampling_frequency 
    10.000000
    [simon@hptablet ~]$ cat /sys/bus/iio/devices/iio\:device0/buffer/enable 
    1
    [simon@hptablet ~]$ cat /sys/bus/iio/devices/iio\:device0/buffer/length 
    128
    [simon@hptablet ~]$ cat /sys/bus/iio/devices/iio\:device0/scan_elements/in_accel_x_en 
    1
    [simon@hptablet ~]$ cat /sys/bus/iio/devices/iio\:device0/scan_elements/in_accel_y_en 
    1
    [simon@hptablet ~]$ cat /sys/bus/iio/devices/iio\:device0/scan_elements/in_accel_z_en 
    1
    
Still no continuous data in /dev/iio:device0. I'm not sure whether I'm missing a flag or the driver is buggy..

frebib, 2016-01-01:
I don't have anything to contribute except to say that the issue is also affecting me, I've been following for a while. cat'ing the devices yields data but only occasionally makes the rotation work. I'm happy to test things as I'd love to see this resolved.

frebib, 2016-01-12:
This seems related. http://www.spinics.net/lists/linux-iio/msg22241.html
It throws out the same errors as mine which is an interesting sign. I'll do some digging when my surface returns

espectalll123, 2016-03-02:
Here I have a Surface Pro 2 with Ubuntu GNOME 16.04, same problem. I only discovered that GNOME could rotate the screen because I accidentaly did _once_. Never worked since.

I reloaded IIO with the `G_MESSAGES_DEBUG=all` flag to see if any error appeared, but the logs that I got didn't seem to show up anything useful:

```
** (process:2821): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:045E:07A9.000D/HID-SENSOR-200041.4.auto/iio:device3
** (process:2821): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:045E:07A9.000D/HID-SENSOR-200041.4.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
** (process:2821): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:045E:07A9.000D/HID-SENSOR-200073.1.auto/iio:device0
** (process:2821): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:045E:07A9.000D/HID-SENSOR-200073.1.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:2821): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:045E:07A9.000D/HID-SENSOR-200083.3.auto/iio:device2
** (process:2821): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:045E:07A9.000D/HID-SENSOR-200083.3.auto/iio:device2 of type compass at IIO Buffer Compass
** (process:2821): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:045E:07A9.000D/HID-SENSOR-200073.1.auto/trigger0
** (process:2821): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2821): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2821): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2821): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2821): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2821): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2821): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:045E:07A9.000D/HID-SENSOR-200041.4.auto/trigger3
** (process:2821): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2821): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2821): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/0003:045E:07A9.000D/HID-SENSOR-200083.3.auto/trigger2
** (process:2821): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2821): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2821): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2821): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2821): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2821): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2821): DEBUG: No new data available
** (process:2821): DEBUG: No new data available
** (process:2821): DEBUG: No new data available
** (process:2821): DEBUG: No new data available
** (process:2821): DEBUG: No new data available
** (process:2821): DEBUG: No new data available
** (process:2821): DEBUG: No new data available
```
...and would keep like that forever. `cat /sys/bus/iio/devices/iio\:device0/in_accel_[axis]_raw`, however, managed to returned me data that seemed to respond to my actions. How awkward...

Bradan, 2016-06-26:
Also happens here on Thinkpad Helix. What driver is this bug inside?

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

closed 2016-11-22 by hadess

smithfred, 2017-06-14:
Still broken with iio-sensor-proxy 2.2.1 and standard F25 kernel 4.11.3 (as of 4.11, none of the above-mentioned other-hardware patches are needed any more).

The workaround of continually polling the accel files in /sys still works.

Is #116 actually supposed to be a workaround for this issue, or is that mistaken?