hadess/iio-sensor-proxy #62: Lenovo Thinkpad Yoga 12

dev-uhuru, 2015-12-10:
I run Ubuntu GNOME 15.10 (upgraded to Gnome 3.18 with staging ppa) and since it seems 3.18 supports screen rotation (and brightness adjustment) although this does not work on my laptop, I followed the different steps in the readme document. 

I'm sorry, I tried to attach the files but it doesn't seem to work (it reads: Something went really wrong, and we can’t process that file.) Please see copy/past below:

node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};

The output of DEBUG is:

** (process:7390): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.7/3-1.7:1.0/0003:0483:91D1.0002/HID-SENSOR-200041.7.auto/iio:device1
** (process:7390): DEBUG: Found device /sys/devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.7/3-1.7:1.0/0003:0483:91D1.0002/HID-SENSOR-200041.7.auto/iio:device1 of type ambient light sensor at IIO Buffer Light sensor
** (process:7390): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.7/3-1.7:1.0/0003:0483:91D1.0002/HID-SENSOR-200073.2.auto/iio:device4
** (process:7390): DEBUG: Found device /sys/devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.7/3-1.7:1.0/0003:0483:91D1.0002/HID-SENSOR-200073.2.auto/iio:device4 of type accelerometer at IIO Buffer accelerometer
** (process:7390): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.7/3-1.7:1.0/0003:0483:91D1.0002/HID-SENSOR-200083.4.auto/iio:device5
** (process:7390): DEBUG: Found device /sys/devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.7/3-1.7:1.0/0003:0483:91D1.0002/HID-SENSOR-200083.4.auto/iio:device5 of type compass at IIO Buffer Compass
** (process:7390): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.7/3-1.7:1.0/0003:0483:91D1.0002/HID-SENSOR-200073.2.auto/trigger4
** (process:7390): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.7/3-1.7:1.0/0003:0483:91D1.0002/HID-SENSOR-200041.7.auto/trigger1
** (process:7390): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.7/3-1.7:1.0/0003:0483:91D1.0002/HID-SENSOR-200083.4.auto/trigger5
** (process:7390): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:7390): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:7390): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:7390): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:7390): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device

Regards,

Thomas

LittleHuba, 2015-12-28:
Have the same problem with the same device. I would really appreciate a fix.

Regards,

LH

LittleHuba, 2016-01-10:
This bug is the same as bug #66. It also works after going into suspend mode.

hadess, 2016-01-12:
Best test with kernel 4.4 as well then.

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

closed 2016-11-22 by hadess