hadess/iio-sensor-proxy #63: gdm errors - not authorized

f3flight, 2015-12-13:
Today I have downloaded, compiled and installed the iio-sensor-proxy service.
It works okay, but what is strange is that every time the light changes (light sensor value), I get the following error message in log:

```
Dec 13 21:41:09 universe pkexec[2887]: gdm: Error executing command as another user: Not authorized [USER=root] [TTY=unknown] [CWD=/var/lib/gdm] [COMMAND=/usr/lib/gnome-settings-daemon/gsd-backlight-helper --set-brightness 927]
Dec 13 21:41:09 universe gnome-session[366]: Error executing command as another user: Not authorized
Dec 13 21:41:09 universe gnome-session[366]: This incident has been reported.
Dec 13 21:41:10 universe pkexec[2891]: gdm: Error executing command as another user: Not authorized [USER=root] [TTY=unknown] [CWD=/var/lib/gdm] [COMMAND=/usr/lib/gnome-settings-daemon/gsd-backlight-helper --set-brightness 927]
Dec 13 21:41:10 universe gnome-session[366]: Error executing command as another user: Not authorized
Dec 13 21:41:10 universe gnome-session[366]: This incident has been reported.
```

The value of --set-brightness is always the same - 927. If I enable automatic brightness then next to these messages I see:

```
Dec 13 21:39:58 universe pkexec[2824]: pam_unix(polkit-1:session): session opened for user root by (uid=1000)
Dec 13 21:39:58 universe pkexec[2824]: user: Executing command [USER=root] [TTY=unknown] [CWD=/home/user] [COMMAND=/usr/lib/gnome-settings-daemon/gsd-backlight-helper --set-brightness 477]
```

Which looks ok and the value here changes. But if I disable automatic brightness the first example still keeps popping in log (each time the light sensor value changes).

Any ideas what is going wrong here, why there is a strange attempt to change brightness to 927? I use Arch Linux, kernel is 3.2.5, gdm & gnome.

f3flight, 2015-12-13:
filed a bug in gnome just in case - https://bugzilla.gnome.org/show_bug.cgi?id=759414

hadess, 2016-01-11:
Upstream bug was fixed.

closed 2016-01-11 by hadess