hadess/iio-sensor-proxy #64: Yoga 900 rotation issues

nacc, 2015-12-17:
Running Gnome 3.18 from Ubuntu's PPA, which should have the rotation support, and w/ the github version of iio-sensor-proxy, I can see the sensors updating in `watch -n 0.1 cat /sys/bus/iio/devices/iio\:device*/*raw*`

But `G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy` seems to not be seeing any of the updates (log attached).

-Nish
[GMESSAGES.txt](https://github.com/hadess/iio-sensor-proxy/files/65898/GMESSAGES.txt)

nacc, 2015-12-17:
Another data point, maybe this is a kernel issue? The corresponding /dev/ nodes are not producing any output, even though the sysfs *raw* files are?

nacc, 2015-12-17:
Ok, in looking into it a bit more, it seems like the Yoga 900's acclerometer is using the hid-sensor-accel-3d driver, which only supports exporting data over the raw file interfaces in sysfs (I think), not via any character devices? I sent an e-mail to LKML and others to see if that's intentional...

hadess, 2016-01-11:
Look for the kernel patch "HID: sensor-hub: Add quirk for Lenovo Yoga 900 with ITE Chips".

Not an iio-sensor-proxy issue.

closed 2016-01-11 by hadess