hadess/iio-sensor-proxy #66: Only works after waking from suspend (Yoga 3 Pro)

redbmk, 2016-01-02:
I was really frustrated trying to get this to work the other day when I first installed it. Using introspect I would get something like this:

```
interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
```

Today I noticed it was working fine, so I was surprised. After a restart though, it stopped working again. I finally figured out that if I close the laptop screen, which puts it into suspend mode, then it will work once I open the screen again. I now get something like this that changes depending on the orientation or light level.

```
 interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'normal';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 289.51999352872372;
  };
```

I'm still curious if this has something to do with closing the screen specifically, or going into suspend mode. Ideally this would work as soon as the computer is started.

redbmk, 2016-01-02:
Sorry, I forgot to mention some useful information. I'm running the latest stable version of the linux kernel:

```
$ uname -r
4.3.3-040303-generic
```

I'm also using Gnome 3.18.0 (on Ubuntu 5.10).

```
$ cat /usr/share/gnome/gnome-version.xml 
<?xml version="1.0" encoding="UTF-8"?>
<gnome-version>
 <platform>3</platform>
 <minor>18</minor>
 <micro>0</micro>
 <distributor>Ubuntu</distributor>
 <date>2015-09-23</date>
</gnome-version>
```

redbmk, 2016-01-04:
A little more information: I tried running in debug mode and got the output below. The first big section of `No new data available`s was before suspending the laptop. During that time I also tried reorienting the device. After it woke back up the data seems more significant. I reoriented a few times and it was behaving properly.

```
** (process:2293): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200041.7.auto/iio:device3
** (process:2293): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200041.7.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
** (process:2293): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200073.2.auto/iio:device1
** (process:2293): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200073.2.auto/iio:device1 of type accelerometer at IIO Buffer accelerometer
** (process:2293): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200083.4.auto/iio:device4
** (process:2293): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200083.4.auto/iio:device4 of type compass at IIO Buffer Compass
** (process:2293): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200073.2.auto/trigger1
** (process:2293): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2293): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2293): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2293): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2293): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2293): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2293): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200041.7.auto/trigger3
** (process:2293): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2293): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2293): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8386.0001/HID-SENSOR-200083.4.auto/trigger4
** (process:2293): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2293): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2293): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2293): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2293): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2293): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2293): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2293): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2293): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2293): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2293): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2293): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: Handling driver refcounting method 'ReleaseLight' for ambient light sensor device
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
[... cut ...]
** (process:2293): DEBUG: Read from IIO: -46, -933, -308
** (process:2293): DEBUG: Accel sent by driver (quirk applied): 0, 9, -3
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: Read from IIO: -46, -937, -328
** (process:2293): DEBUG: Accel sent by driver (quirk applied): 0, 9, -3
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
** (process:2293): DEBUG: No new data available
```

badshah400, 2016-01-11:
Also works if I suspend by using the power button and leaving the screen as is, i.e., it doesn't seem tied specifically to closing the screen, but rather to the suspending and waking up instead.

hadess, 2016-01-12:
You'll want to test with kernel 4.4

badshah400, 2016-01-12:
uname -r: 4.4.0-rc8-1.ge628e30-default
Same result, i.e. doesn't work right off the boot, but works after waking from suspend and onwards.

badshah400, 2016-01-16:
So yeah, just to confirm, doesn't work with kernel 4.4.0 final either, not unless I wake up from suspend at least once after booting.

andreasn, 2016-01-23:
I have the exact same behavior on Fedora 23 (but that's a lot of progress from the previous state!)
This laptop doesn't have anything critical on it, so I think I'll wipe it tomorrow and try if it works better in rawhide.

andreasn, 2016-01-23:
Unfortunately doesn't work any better with a 4.5-kernel.
4.5.0-0.rc0.git8.1.fc24.x86_64

andreasn, 2016-01-26:
I've filed a kernel bug about this now
https://bugzilla.kernel.org/show_bug.cgi?id=111321

eduncan911, 2016-01-31:
Subscribing... As my Lenovo ThinkPad Helix acts the same way (rotation doesn't work, until I suspend and wake it back up).

Ambient Light isn't working/detecting at all... May be something else.  Still debugging that one.

F4FXL, 2016-02-03:
Hi,

I have a Similar issue on Yoga 2 Pro : https://bugzilla.kernel.org/show_bug.cgi?id=109501

f3flight, 2016-02-10:
Arch Linux, Dell Venue 11 Pro 7140 - same issue on default kernel 4.4.1. Downgrading to 4.1.7-lts fixes it.

fajabird, 2016-02-16:
Arch Linux, Dell Venue 11 7130 vPro (Core i5) - same issue, only works after suspend/resume. Tested both, latest git-version and stable 1.1.1.

diraimondo, 2016-02-26:
The same here but with a USB ColorHug ALS sensor on a Dell XPS 13 9350 running Arch Linux using iio-sensor-proxy v.1.1.1 or latest git-version. The sensor starts to work only after a suspend-resume cycle. I'm using kernel 4.4 but tested also 4.5rc5.

eages, 2016-04-11:
Can confirm this is a kernel issue for the Thinkpad Yoga S1. Running Fedora 23, kernel ver 4.2.3 works as expected but kernel 4.4.6 and higher doesn't.

chadotter, 2016-04-17:
Same issue for the Dell XPS 12 9Q33 running Ubuntu 16.04 RC with kernel 4.4.0-18-generic.

kristijansuler, 2016-04-25:
Same issue for Asus Zenbook UX305LA running Ubuntu GNOME 16.04 LTS with kernel 4.4.0-21-generic

bgaff, 2016-05-19:
Same issue Dell Inspiron 13 7000, 4.4.0-22-generic

nacc, 2016-07-05:
I was/am seeing this with the Ubuntu 16.04 kernel, but suspend&resume has no impact on my system (Yoga 900, at least in 3 attempts yesterday). I believe it did have an impact before (resulting in working sensors), so possibly this is a regression. I just built and booted into 4.7-rc5+ (0b295dd5b8eb755bc7b57220d93705994090e954 is my current tip HEAD) and screen rotation worked fine, regardless of suspend & resume, though, so that's progress! I'm not sure I'll have time to bisect this, but maybe someone else can (and at least for Ubuntu, if you can let me know the commit, I'll work on getting it fixed in the 16.04 kernel).

molecular, 2016-09-06:
same as @nacc here:
 * yoga 900
 * kernel 4.4.0
 * workaround (suspend) does *not* help

eages, 2016-09-08:
I can confirm the sensor's working again in kernel 4.7.2 on the Thinkpad Yoga S1. No suspend required.

badshah400, 2016-09-08:
Thinkpad Yoga 260 (Intel Sensor Hub patches applied): still doesn't work with Kernel 4.7.2 from a cold boot. Works after suspend.

nacc, 2016-09-22:
FYI, Ubuntu 16.10's latest kernels (4.8.0-14) with iio-sensor-proxy from the Ubuntu archive leads to working rotation immediately.

brokkr, 2016-10-29:
Lenovo Yoga 2 Pro here. 

With regards to Ubuntu 16.10: I just booted up the live media and as far as I can see, it's a regression. monitor-sensor just says 'waiting for iio-sensor-proxy'. On my current install (16.04) the prior checks described work and so monitor-sensor merely stays quiet until waking from suspend (after which it starts reporting correctly).

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

closed 2016-11-22 by hadess