hadess/iio-sensor-proxy #69: stopped working for some reason

f3flight, 2016-02-07:
I have built iio-sensor-proxy in Arch Linux using PKGBUILD - initially it worked right after installing the resulting package, but after reboot looks like it does not work anymore (not sure whether it was a reboot or something else, but eventually I noticed that screen rotation does not work anymore and monitor-sensor shows nothing).
Debugging dbus output attached - basically says "No new data available".
[sensor-proxy-out.txt](https://github.com/hadess/iio-sensor-proxy/files/120844/sensor-proxy-out.txt)
[udevadm.txt](https://github.com/hadess/iio-sensor-proxy/files/120847/udevadm.txt)
[gdbus.txt](https://github.com/hadess/iio-sensor-proxy/files/120848/gdbus.txt)
should there be a kernel module loaded for this to work? Though I don't remember doing anything special with the kernel... Let me know if I can get any additional info to debug this issue.

frebib, 2016-02-08:
This issue is coming up more and more it seems. I get the impression it's a kernel bug.. 
What device are you running on?

f3flight, 2016-02-08:
Sad to hear! My device is Dell Venue 11 Pro 7140.

f3flight, 2016-02-10:
Latest finding - I have the same issue as #66 - the sensors start working after suspend (I use "freeze") and resume. Sensors do not work initially after boot. Kernel 4.4.1. Probably you can merge / close this issue. I will try using an LTS kernel, will update in #66.

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

closed 2016-11-22 by hadess