hadess/iio-sensor-proxy #71: Lenovo Thinkpad X1 Yoga

jbajor, 2016-02-25:
Unfortunately, rotation and ambient light sensor don't work on Lenovo Thinkpad X1 Yoga. Running Fedora 23, Gnome 3.18.3, kernel 4.5.0-0.rc4

```
[root@bender]/home/jacek# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:4570): DEBUG: Found input accel at /sys/devices/virtual/input/input20/event15
** (process:4570): DEBUG: Found device /sys/devices/virtual/input/input20/event15 of type accelerometer at Input accelerometer
** (process:4570): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:4570): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:4570): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:4570): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device

```

beralt85, 2016-04-12:
Same here. 
According to http://entrenchant.blogspot.lu/2016/02/dispatches-from-x1-front.html:

> The Acer BMA150 accelerometer is detected as a joystick device, so various /sys/bus/iio -based screen orientation scripts like thinkpad-yoga-scripts won't work.

```
[root@azurescens beralt]# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:24940): DEBUG: Found input accel at /sys/devices/virtual/input/input8/event6
** (process:24940): DEBUG: Found device /sys/devices/virtual/input/input8/event6 of type accelerometer at Input accelerometer
** (process:24940): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:24940): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:24940): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:24940): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
```

hadess, 2016-04-19:
You need to run `monitor-sensor` or something in the session for the data to start reading. Any more output than that?

beralt85, 2016-04-20:
Hello,
thank you so much for looking into that issue. `udevadm info --export-db | grep -C 3 iio` shows only the following entries, but does not show any iio-devices:
```
P: /devices/platform/coretemp.0/hwmon/hwmon2

E: DEVPATH=/devices/platform/coretemp.0/hwmon/hwmon2
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=1651649

--
P: /devices/platform/thinkpad_hwmon/hwmon/hwmon0
E: DEVPATH=/devices/platform/thinkpad_hwmon/hwmon/hwmon0
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=1601445

--
P: /devices/virtual/hwmon/hwmon1
E: DEVPATH=/devices/virtual/hwmon/hwmon1
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=1619159
```
However, scanning for info on the accelerometer  with `udevadm info --export-db | grep -C 7 accel` gives:
```
P: /devices/virtual/input/input8
E: ABS=7
E: DEVPATH=/devices/virtual/input/input8
E: EV=9
E: ID_INPUT=1
E: ID_INPUT_ACCELEROMETER=1
E: MODALIAS=input:b0019v0000p0000e0000-e0,3,kra0,1,2,mlsfw
E: NAME="Acer BMA150 accelerometer"
E: PHYS="wmi/input1"
E: PRODUCT=19/0/0/0
E: PROP=0
E: SUBSYSTEM=input
E: TAGS=:seat:
E: USEC_INITIALIZED=1649977
```
This is probably what was referred to in the link in my previous comment.

Executing `gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy` results in
```
Error: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files
```

`monitor-sensor` starts without any output.

Note that I installed via an Archlinux PKGBuild from the AUR (https://aur.archlinux.org/packages/iio-sensor-proxy/). For testing, I also installed from your git version. (Btw, I had some problems doing so because your installation method via autogen.sh script uses different filepaths. Also the installation instructions using `./configure && make && make install` do not work anymore.)

Do you think that one can easily change your programm to work with this acceleration sensor?

Tell me, if I can provide any more information.

adrinjalali, 2016-04-30:
Hi,

Using your latest code, this is what I get:

```
$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = false;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};

```

And probably as suspected from `      readonly s AccelerometerOrientation = 'undefined';`, rotation doesn't work, even after calling `monitor-sensor`, which results in this output:

```
$ monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
** Message: === Has accelerometer (orientation: undefined)
** Message: === No ambient light sensor

```

I'd be more than happy to help. Please let me know what I can do.

hadess, 2016-04-30:
What's the output of iio-sensor-proxy as you posted in your original comment?

adrinjalali, 2016-05-01:
Here's the output of `iio-sensor-proxy`:

```
# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:29188): DEBUG: Found input accel at /sys/devices/virtual/input/input8/event6
** (process:29188): DEBUG: Found device /sys/devices/virtual/input/input8/event6 of type accelerometer at Input accelerometer
** (process:29188): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:29188): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
```

And when I run `monitor-sensor`, these extra two lines appear in the output of `iio-sensor-proxy`:
```
** (process:29188): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:29188): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
```

Not sure if it helps, but when `iio-sensor-proxy` is running, the corresponding *rotate* button in Gnome appears. Another fact, that the output of this command is empty:
`# find /sys -name "*iio*"`
But I'm not sure if it's relevant here.

beralt85, 2016-05-04:
Even installing your new version I get
```
Error: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files
```

upon calling `gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy`

The "lock rotation" butoon appears in gnome shell, when iio-sensor-proxy is running

sils1297, 2016-05-24:
@beralt85 `systemctl start iio-sensor-proxy.service` solves the issue you have for me.

I do get the same "undefined" accelerometer orientation as @adrinjalali

Soulwest, 2016-06-15:
Same issue

volfyd, 2016-07-05:
I have the Thinkpad X1 tablet (May 2016 edition) and I am having the same issue with a BMA150 accelerometer being detected by the acer_wmi driver but nothing useful happening with iio-sensor-proxy.

volfyd, 2016-07-05:
I wonder if this is relevant: https://lkml.org/lkml/2016/5/31/3

turbovan, 2016-07-08:
I also have the same problems with an X1 Yoga running Arch with Gnome 3.20 and the AUR package of iio-sensor-proxy. Can assist with testing

omar-polo, 2016-07-11:
Hi!

I had a similar issue. I'm on Lenovo Yoga 3 Pro running arch linux (linux version: `4.6.3-1-ARCH`). I've installed the `iio-sensor-proxy-git` from AUR and, when I run `gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy` I got:

```
Error: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name net.hadess.SensorProxy was not provided by any .service files
```

I think it could be relevant: unistalling that package and installing `iio-sensor-proxy` (still from AUR) solved the issue for me

rannbaron, 2016-07-15:
I have progress to report! I installed Srinavas Pandruvada's patches (http://thread.gmane.org/gmane.linux.kernel/2250222) against kernel 4.7-rc4 on Debian, and edited the iio-sensor-proxy.service file. While the screen did not initially rotate, before or after a suspend, I was poking around and discovered that if I run 

cat /sys/bus/iio/devices/iio\:device3/in_accel_x_raw

as root while the screen is physically rotated, I get rotation!  If I execute the same command again after returning the screen to its original orientation, it rotates back!

I am in direct contact with Srinavas and will update here if/when I hear back from him regarding this progress.

rannbaron, 2016-07-15:
Also:

$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'normal';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 107.65200511319563;
  };
  node Compass {
  };
};

This does not change upon rotation, but does once I execute the command in the previous comment!

rannbaron, 2016-07-16:
With the patches at https://lkml.org/lkml/2016/7/5/235, plus the addition of the line 

After=multi-user.target

to the [Unit] section of the iio-sensor-proxy.service file, I can confirm that screen rotation works. It's not entirely smooth, but it certainly works.

Yay!

beralt85, 2016-07-26:
@rannbaron That's great news. However, I don't feel quite comfortable patching the kernel myself (I don't really know what to do. Would I have to copy all the diffs out of the newsgroup, and then manually apply those to the kernel sources (ie for kernel 4.7-rc5)? Sorry, I am not that deep into all that...)

Do you know where I can find a patched version of the kernel sources?
(Or even better, find out if they were implemented already in some kernel in the arch user repository). I would really like to help out with the testing...

rannbaron, 2016-07-27:
@beralt85 You can download the patches indeed, then manually apply to kernel sources, following the documentation there. I got a bit of help from the developer, as there were some kernel options to be switched on as well. Email him. He's really nice.

If you're running Arch, you really should give compiling your own kernel a go. It's not that hard, and it's good experience.

I run Debian, so I have no idea what's available in the Arch repositories. Maybe if you ask nicely on the forums someone will do it for you.  I think it'll be at least kernel 4.9 before this makes it into mainline, unfortunately.

grensjo, 2016-07-31:
@rannbaron Could you share which kernel options you had to switch on? I have managed to patch, compile and install the kernel, but nothing seems to have changed, so there is probably something I have missed.

I run Arch, so if I manage to get the patches working I could create an AUR package for this.

rannbaron, 2016-07-31:
@grensjo Here you go:

CONFIG_HID_SENSOR_HUB=m
CONFIG_HID_SENSOR_CUSTOM_SENSOR=m
CONFIG_HID_SENSOR_ACCEL_3D=m
CONFIG_HID_SENSOR_IIO_COMMON=m
CONFIG_HID_SENSOR_IIO_TRIGGER=m
CONFIG_HID_SENSOR_GYRO_3D=m
CONFIG_HID_SENSOR_ALS=m
CONFIG_HID_SENSOR_PROX=m
CONFIG_HID_SENSOR_MAGNETOMETER_3D=m
CONFIG_HID_SENSOR_INCLINOMETER_3D=m
CONFIG_HID_SENSOR_DEVICE_ROTATION=m
CONFIG_HID_SENSOR_PRESS=m
CONFIG_INTEL_ISH_HID=y

ids1024, 2016-08-25:
> With the patches at https://lkml.org/lkml/2016/7/5/235, plus the addition of the line
>
> After=multi-user.target
>
> to the [Unit] section of the iio-sensor-proxy.service file, I can confirm that screen rotation works. It's not entirely smooth, but it certainly works.

I can confirm that this works on my Lenovo Yoga 710.

Perhaps the `After=multi-user.target` line should be added to the official .service file in the repo? Perhaps it isn't needed on other systems, but I presume it would do no harm.

rannbaron, 2016-09-08:
Good news folks: It seems that this patch will be in kernel 4.9 in a few months, unless someone on the kernel team objects to it for some reason.

sarmbruster, 2016-10-27:
On Ubuntu 16.04 with Mainline Kernel 4.9-rc2 I get luckily get now some output from `monitor-sensor`:
``` monitor-sensor 
** Message: Accelerometer orientation changed: right-up
** Message: Light changed: 173.235008 (lux)
** Message: Light changed: 47.364002 (lux)
** Message: Light changed: 46.715002 (lux)
** Message: Light changed: 16.220001 (lux)
** Message: Light changed: 10.381000 (lux)
** Message: Light changed: 9.991000 (lux)
** Message: Light changed: 8.045000 (lux)
** Message: Accelerometer orientation changed: normal
** Message: Light changed: 15.182001 (lux)
** Message: Light changed: 15.442001 (lux)
** Message: Light changed: 15.961001 (lux)
** Message: Light changed: 15.701001 (lux)
** Message: Light changed: 15.442001 (lux)
** Message: Light changed: 15.182001 (lux)
** Message: Accelerometer orientation changed: right-up
** Message: Light changed: 12.587001 (lux)
** Message: Light changed: 9.862000 (lux)
** Message: Light changed: 10.510000 (lux)
** Message: Light changed: 10.640001 (lux)
** Message: Light changed: 10.770001 (lux)
** Message: Light changed: 10.640001 (lux)
** Message: Accelerometer orientation changed: normal
** Message: Light changed: 14.144001 (lux)
** Message: Light changed: 14.793001 (lux)
** Message: Accelerometer orientation changed: right-up
** Message: Light changed: 10.640001 (lux)
** Message: Light changed: 9.472000 (lux)
```

hadess, 2016-11-22:
If this is working, I'll mark this as closed. If the problem persists, please make sure to file separate bugs for your problems.

commit 51b35aa3 (2016-04-20): data: Run udev rules later

40-* is too early to have some of the tags added to devices we want to
use. For example, input accelerometers wouldn't be tagged for
iio-sensor-proxy usage, as the tagging would be done in 50-*.

80 seems to be a good priority.

Closes #71
closed 2016-11-22 by hadess