hadess/iio-sensor-proxy #72: Lenovo 260 yoga does not recognize sensors on Ubuntu 16.04

mabawsa, 2016-03-04:
Happens both on the repo and git versions
root@yoga:~# uname -r
4.5.0-999-generic

root@yoga:~# udevadm info --export-db > iio.txt (relevant)
P: /devices/platform/coretemp.0/hwmon/hwmon2
E: DEVPATH=/devices/platform/coretemp.0/hwmon/hwmon2
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=3700456
P: /devices/platform/thinkpad_hwmon/hwmon/hwmon1
E: DEVPATH=/devices/platform/thinkpad_hwmon/hwmon/hwmon1
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=3670994
P: /devices/virtual/hwmon/hwmon0
E: DEVPATH=/devices/virtual/hwmon/hwmon0
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=3632239

root@yoga:~# systemctl status iio-sensor-proxy.service● iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/lib/systemd/system/iio-sensor-proxy.service; static; vendor 
   Active: inactive (dead) since 土 2016-03-05 00:08:43 JST; 8min ago
  Process: 2845 ExecStart=/usr/sbin/iio-sensor-proxy (code=exited, status=0/SUCC
 Main PID: 2845 (code=exited, status=0/SUCCESS)

 3月 05 00:08:42 yoga systemd[1]: Stopped IIO Sensor Proxy service.
 3月 05 00:08:42 yoga systemd[1]: Starting IIO Sensor Proxy service...
 3月 05 00:08:43 yoga systemd[1]: Started IIO Sensor Proxy service.

root@yoga:~# G_MESSAGES_DEBUG=all iio-sensor-proxy
** (process:3743): DEBUG: Could not find any supported sensors

hadess, 2016-03-04:
You have two hwmon devices (it's a type of hardware monitor interface in the Linux kernel, used by many different driver types, and in particular, for brightness sensors on Macs and Samsung laptops).

```
P: /devices/platform/coretemp.0/hwmon/hwmon2
E: DEVPATH=/devices/platform/coretemp.0/hwmon/hwmon2
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=3700456
```

This is a temperature sensor.

```
P: /devices/platform/thinkpad_hwmon/hwmon/hwmon1
E: DEVPATH=/devices/platform/thinkpad_hwmon/hwmon/hwmon1
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=3670994
```

This is an interface to access the fan settings for the laptop.

```
P: /devices/virtual/hwmon/hwmon0
E: DEVPATH=/devices/virtual/hwmon/hwmon0
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=3632239
```

This is a virtual sensor attached to the first one at the top.

In short, your laptop has no sensors or your kernel doesn't know how to support them. I'm afraid that there's nothing iio-sensor-proxy can do if the kernel doesn't support it.

Please report it to your distribution to verify which drivers are enabled, and investigate what sort of sensor(s) are available in your laptop.

closed 2016-03-04 by hadess

monokrome, 2017-01-27:
@mabawsa Same issue here. Did you find any solution that others may find interesting?

mabawsa, 2017-01-27:
It works with kernel 4.9 and upwards and the latest iio-sensor proxy. 
The novelty soon wore off for me though.