hadess/iio-sensor-proxy #73: Fails to get status from Accelerometer

pksadiq, 2016-03-23:
iio-sensor-proxy detects an accelerometer on my Laptop, though I think its the one on my Hard disk.
The data read by the iio-sensor-proxy is shown as 'undefined'. It shall be nice to support such accelerometers.

Output of gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy :
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = false;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};

Attaching the output of "udevadm info --export-db"
[udevadm-info.txt](https://github.com/hadess/iio-sensor-proxy/files/185956/udevadm-info.txt)

hadess, 2016-04-19:
Actually, we don't want to support those accelerometers. People get confused by the icons in gnome-shell that are more directly associated to tablets or convertibles.

pksadiq, 2016-04-19:
though "readonly s AccelerometerOrientation" is "undefined" gnome-shell shows the screen rotation lock.  gnome-shell seems not checking if its undefined or not.

May be its better to list only supported (whose orientation data can be extracted) devices by iio-proxy-sensor, unless explicitly requested for every device, so that gnome-shell won't get confused. I don't know if this change would create an API break, may be worth.

hadess, 2016-04-19:
> though "readonly s AccelerometerOrientation" is "undefined" gnome-shell shows the screen rotation > lock. gnome-shell seems not checking if its undefined or not.

Because it doesn't expect iio-sensor-proxy bugs. And I'd rather fix iio-sensor-proxy than add special cases in its users.

pksadiq, 2016-04-19:
> Because it doesn't expect iio-sensor-proxy bugs. And I'd rather fix
> iio-sensor-proxy than add special cases in its users.
>
Thanks for looking into this.

hadess, 2016-04-19:
In aa9476aae54304b6f54c0adeb505ddb26b5a825b, I added support for polling those accelerometers, so your laptop, with a newer version of iio-sensor-proxy, should be able to rotate.

pksadiq, 2016-04-19:
I built iio-sensor-proxy from master, installed, and restarted my system. I found no difference:

gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy:

node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = false;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};

Its "undefined". But I get the rotation lock button in gnome-shell.
I am missing anything?
How can I get the version of iio-sensor-proxy being used by gnome-shell/system?
monitor-sensor --version or --help does nothing. Nor with iio-sensor-proxy command.

hadess, 2016-04-19:
Attach the output from iio-sensor-proxy, as explained in the README, please.

pksadiq, 2016-04-20:
udevadm info --export-db | grep iio -A3 -B4:

P: /devices/platform/coretemp.0/hwmon/hwmon0
E: DEVPATH=/devices/platform/coretemp.0/hwmon/hwmon0
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=20356886

G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy:

** (process:2190): DEBUG: Found input accel at /sys/devices/virtual/input/input13/event12
** (process:2190): DEBUG: Found device /sys/devices/virtual/input/input13/event12 of type accelerometer at Input accelerometer
** (process:2190): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2190): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2190): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2190): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device

hadess, 2016-04-20:
The udevadm output if useless, it's for a different device which we don't support.

Please try running iio-sensor-proxy under gdb, and Ctrl+C it after a while. We'll see if it's stuck.

pksadiq, 2016-04-20:
On 4/20/16, hadess <notifications@github.com> wrote:
> The udevadm output if useless, it's for a different device which we don't
> support.
>
> Please try running iio-sensor-proxy under gdb, and Ctrl+C it after a while.
> We'll see if it's stuck.
>

Output in gdb (run as root):

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ffff4a52700 (LWP 17275)]
[New Thread 0x7fffeffff700 (LWP 17276)]
[New Thread 0x7fffef7fe700 (LWP 17277)]
[Thread 0x7fffeffff700 (LWP 17276) exited]
^C
Program received signal SIGINT, Interrupt.
0x00007ffff6cf9fdd in poll () from /lib64/libc.so.6

I didn't find a make debug target for iio-sensor-proxy, or something
like ./configure --enable-debug. It shall be better  to have one.

hadess, 2016-11-23:
You need to run "bt" at the gdb prompt to see the backtrace. Looks like the kernel driver isn't giving any useful information in this case. You can also test with `evtest` and see whether any data is output by the device.

If everything went well, you should see a loop of "Accel read from input on...". If not, either it cannot open the device associated, or it's getting stuck reading it.

hadess, 2016-11-23:
After debugging this on IRC, we could not make the accelerometer appear in the initial device list of evtest. The device is still detected by iio-sensor-proxy, but I'm guessing that it should start throwing errors, since c1bc67339bcbd8a6d273d300549883fddbc1a926 added warnings when the device node could not be opened.

I've asked @pksadiq to send a mail to the maintainers of the driver, as listed by the get_maintainers script in the kernel. @garnacho had the same problem on one of his devices, but did not investigate further.

Please reopen when the kernel driver shows up in evtest properly, and we can get events from it. Thanks for your time trying to debug this.

closed 2016-11-23 by hadess