hadess/iio-sensor-proxy #74: Thinkpad Yoga S1 / Accelerometer gets wrong value for a short period of time / Stopped working

mairin, 2016-03-30:
Hi hadess,

I talked to you about https://bugzilla.gnome.org/show_bug.cgi?id=763751 a couple of weeks ago.

Since then I had the rotate disabled, and when I have tried to turn off the rotate lock to get it to rotate again it no longer rotates. When I follow the instructions in https://github.com/hadess/iio-sensor-proxy/blob/master/README.md, everything looks good up until I run monitor-sensor, and then I get no output. 

Am I doing something wrong? I wanted to attach the debug output as directed in the GNOME bug but I can't get it to rotate anymore so I can't reproduce the issue to get the output :(

hadess, 2016-04-19:
If monitor-sensor doesn't show any updates, that's probably because iio-sensor-proxy isn't sending data. In the README, there's some explanation on how to run iio-sensor-proxy with debug enabled. If you see constant "No data to read" or somesuch, then it's because the kernel broke :/

hadess, 2016-11-22:
Does this work now?

closed 2016-12-12 by hadess