hadess/iio-sensor-proxy #75: No sensors data on hp spectre x360

andreacampanella, 2016-03-31:
iio-sensor-proxy can find the sensors but monitor-sensor doen't output anything:

** (process:3284): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8350.0004/HID-SENSOR-200073.2.auto/iio:device3
** (process:3284): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8350.0004/HID-SENSOR-200073.2.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:3284): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8350.0004/HID-SENSOR-200083.4.auto/iio:device4
** (process:3284): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8350.0004/HID-SENSOR-200083.4.auto/iio:device4 of type compass at IIO Buffer Compass
** (process:3284): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8350.0004/HID-SENSOR-200073.2.auto/trigger3
** (process:3284): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3284): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3284): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3284): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3284): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3284): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3284): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8350.0004/HID-SENSOR-200083.4.auto/trigger4
** (process:3284): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3284): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3284): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3284): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3284): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3284): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3284): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3284): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3284): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:3284): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:3284): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3284): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available
** (process:3284): DEBUG: No new data available

andreacampanella, 2016-04-25:
uh, any update about this? could be help have more debug/data?

hadess, 2016-04-25:
No, somebody needs to fix the kernel driver.

andreacampanella, 2016-04-25:
What's wrong with it? I can take a look

hadess, 2016-04-25:
It doesn't send data. Look at the other issues with the "driver bug" tag. Some people have mentioned that their sensors start working after a suspend/resume cycle.

andreacampanella, 2016-04-25:
Odd because I can actually read the data form the /sys/ files...

hadess, 2016-04-25:
But the device itself keeps saying that there is no new data. Which is why it's a bug :)

andreacampanella, 2016-04-26:
ok, I will try to find out the maintainer of the driver and report the bug, I guess is the specific module of my sensor hub, right?

hadess, 2016-04-26:
That or the HID sub-system.

andreacampanella, 2016-05-10:
So, I found the bug, the X360 have this sensors under /sys/bus/iio/devices : 

iio:device0  iio:device1  iio:device2  iio:device3  iio:device4 

and their triggers, out of these I can read raw data from 0 (inclinometer) 1 (accellerometer) and 2 (gyroscope) but the 3rd is the quaternion , this always report 0 0 0 0 . 

the 4th have rotation and magnetometer (?) , I can read from these without any problem...

andreacampanella, 2016-05-14:
It did work once. just once, then it did stop , at the moment the sensor are still sending data out, but iio-sensor-proxy is not reading them, it is not a driver problem, this is clearly an implementation problem.

FearlessSpiff, 2016-07-15:
is this fixed? No replies since two month...

hadess, 2016-07-18:
> is this fixed? No replies since two month...

Send me hardware and I'll get to it much quicker.

andreacampanella, 2016-07-18:
How? is a driver issue, right?

hadess, 2016-07-18:
It's a driver issue, and I don't have hardware to root cause it.

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

closed 2016-11-22 by hadess