hadess/iio-sensor-proxy #76: Tested on asus Zenbook ux31a

olerem, 2016-04-01:
Hi,

it is working with acpi-als sensor on Asus Zenbook ux31a,
This patch is needed to enable the sensor:
http://article.gmane.org/gmane.linux.drivers.platform.x86.devel/8975

hadess, 2016-04-19:
It looks like the patch still hasn't been merged. Can you let me know when it is, and I'll update the README. Thanks!

olerem, 2016-04-25:
ok,
i have colormunki photo with light measurement. Should i compare values provided by ALS or it makes no difference?

hadess, 2016-04-25:
Make a difference to what? We just expose the readings from the sensor. If the driver doesn't read them properly, you should report that to the driver maintainers. It will be make no difference to the way most desktops use the sensor.

olerem, 2016-06-06:
The patch is upstream: https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=aca234f6378864d85514be558746c0ea6eabfa8e

asankamal, 2016-09-25:
Asus zenbook UX305, kernel 4.7
It shows sensor output correctly with "monitor-sensor" command.
Gnome 3.20 recognizes and adds the option to enable/disable automatic brightness.
However it doesn't adjust the brightness according to light condition, when enabled

commit 04741e53 (2016-11-23): README: Add Asus Zenbooks to the tested list

Closes: #76
closed 2016-11-23 by hadess