hadess/iio-sensor-proxy #77: Please, test it on Lenovo Flex 2 and Lenovo Yoga 3 13

deadwolfst, 2016-04-20:
Please, test it on Lenovo Flex 2 and Lenovo Yoga 3 13.
Lenovo Flex 2 has accelerometer, but 'G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy' says
** (process:5833): DEBUG: Could not find any supported sensors
By default gnome automatically changes desktop on screen flip. It is strange.

hadess, 2016-04-20:
Feel free to send me hardware if you want me to test it...

Otherwise, did you run it as root, or as a normal user?

deadwolfst, 2016-04-20:
I run it as root

hadess, 2016-04-20:
Attach the udevadm command mentioned in the README.

deadwolfst, 2016-04-20:
there are only these two mentions of iio:
P: /devices/platform/coretemp.0/hwmon/hwmon1
E: DEVPATH=/devices/platform/coretemp.0/hwmon/hwmon1
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=11101128

P: /devices/virtual/hwmon/hwmon0
E: DEVPATH=/devices/virtual/hwmon/hwmon0
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=10303574

hadess, 2016-04-20:
The full output please.

deadwolfst, 2016-04-20:
$ udevadm info --export-db
P: /devices/LNXSYSTM:00
E: DEVPATH=/devices/LNXSYSTM:00
E: MODALIAS=acpi:LNXSYSTM:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/ICV0A12:00
E: DEVPATH=/devices/LNXSYSTM:00/ICV0A12:00
E: ID_VENDOR_FROM_DATABASE=Inside Contactless
E: MODALIAS=acpi:ICV0A12:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=9737179

P: /devices/LNXSYSTM:00/INT33D0:00
E: DEVPATH=/devices/LNXSYSTM:00/INT33D0:00
E: ID_VENDOR_FROM_DATABASE=Interphase Corporation
E: MODALIAS=acpi:INT33D0:PNP0C02:
E: SUBSYSTEM=acpi
E: USEC_INITIALIZED=9737257

P: /devices/LNXSYSTM:00/LNXCPU:00
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:00
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/LNXCPU:01
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:01
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/LNXCPU:02
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:02
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/LNXCPU:03
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:03
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/LNXCPU:04
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:04
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/LNXCPU:05
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:05
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/LNXCPU:06
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:06
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/LNXCPU:07
E: DEVPATH=/devices/LNXSYSTM:00/LNXCPU:07
E: MODALIAS=acpi:LNXCPU:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/LNXPWRBN:00
E: DEVPATH=/devices/LNXSYSTM:00/LNXPWRBN:00
E: DRIVER=button
E: MODALIAS=acpi:LNXPWRBN:
E: SUBSYSTEM=acpi

P: /devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
E: DEVPATH=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
E: EV=3
E: ID_FOR_SEAT=input-acpi-LNXPWRBN_00
[... 5887 lines cut ...]
E: SUBSYSTEM=wmi

P: /devices/virtual/wmi/431F16ED-0C2B-444C-B267-27DEB140CF9C
E: DEVPATH=/devices/virtual/wmi/431F16ED-0C2B-444C-B267-27DEB140CF9C
E: MODALIAS=wmi:431F16ED-0C2B-444C-B267-27DEB140CF9C
E: SUBSYSTEM=wmi

P: /devices/virtual/wmi/ABBC0F6F-8EA1-11D1-00A0-C90629100000
E: DEVPATH=/devices/virtual/wmi/ABBC0F6F-8EA1-11D1-00A0-C90629100000
E: MODALIAS=wmi:ABBC0F6F-8EA1-11D1-00A0-C90629100000
E: SUBSYSTEM=wmi

P: /devices/virtual/wmi/ABBC0F72-8EA1-11D1-00A0-C90629100000
E: DEVPATH=/devices/virtual/wmi/ABBC0F72-8EA1-11D1-00A0-C90629100000
E: MODALIAS=wmi:ABBC0F72-8EA1-11D1-00A0-C90629100000
E: SUBSYSTEM=wmi

P: /devices/virtual/wmi/D9F41781-F633-4400-9355-601770BEC510
E: DEVPATH=/devices/virtual/wmi/D9F41781-F633-4400-9355-601770BEC510
E: MODALIAS=wmi:D9F41781-F633-4400-9355-601770BEC510
E: SUBSYSTEM=wmi

P: /devices/virtual/workqueue/writeback
E: DEVPATH=/devices/virtual/workqueue/writeback
E: SUBSYSTEM=workqueue

hadess, 2016-04-20:
Your kernel doesn't have the right drivers, or your device doesn't have accelerometers. Report the problem to your distribution please, there's nothing that iio-sensor-proxy can do.

closed 2016-04-20 by hadess