hadess/iio-sensor-proxy #79: Automatic screen brightness in Ubuntu Unity with Xenial?

FMuro, 2016-04-21:
First of all, I'd like to thank you for your work. I've installed iio-sensor-proxy in Xenial on a macbook pro 8,2 and the light sensors are detected, I've check this with monitor-sensor. However, the screen brightness or keyboard backlight don't change. I think that Unity in Xenial is mostly Gnome 3.18, so I was expecting this to work. Is there anything I could do to fix this? BTW, it worked in Fedora 23 on the same computed, which I've been using for a while, but have abandoned because it consumes a lot of memory for no apparent reason.

hadess, 2016-04-22:
> First of all, I'd like to thank you for your work. I've installed iio-sensor-proxy in Xenial on a macbook pro 8,2 and the light sensors are detected, I've check this with monitor-sensor. However, [...]

GNOME 3.18 should be new enough, please gather debug output as per README.md

> BTW, it worked in Fedora 23 on the same computed, which I've been using for a while, but have abandoned because it consumes a lot of memory for no apparent reason.

That's most likely incorrect.

FMuro, 2016-04-22:
Thanks for your answer, hadess! Here you are:

** (process:3185): DEBUG: Found IIO poll light at /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:1a/ACPI0008:00/iio:device0
** (process:3185): DEBUG: Found device /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:1a/ACPI0008:00/iio:device0 of type ambient light sensor at IIO Polling Light sensor

BTW, the Fedora thing may well be that I don't remember be, but it might also be due to other things, such as the fact that I used the Radeon card with it, instead of the integrated intel card.

hadess, 2016-04-22:
is that it? Can you run "monitor-sensor" in a separate window/shell as well?

FMuro, 2016-04-22:
I'm afraid I run the debugging command in a constant light intensity. Now, I've approached a lantern to the light sensor so that you can see the variation. I've uploaded two screenshots. One of the is the usual output of monitor-sensor with iio-sensor-proxy running. The other one is the result of your debugging instructions. I hope this is what you meant. 

[image]
[image]

BudgieInWA, 2016-06-18:
I am in the same boat (on a Samsung ATIV 700T, Ubuntu 16.04, 

```
$ systemctl stop iio-sensor-proxy.service
$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:3460): DEBUG: Found IIO poll light at /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:00/PNP0C09:00/ACPI0008:00/iio:device0
** (process:3460): DEBUG: Found device /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:00/PNP0C09:00/ACPI0008:00/iio:device0 of type ambient light sensor at IIO Polling Light sensor
** (process:3460): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.2.auto/iio:device3
** (process:3460): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.2.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:3460): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.4.auto/iio:device4
** (process:3460): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.4.auto/iio:device4 of type compass at IIO Buffer Compass
** (process:3460): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.2.auto/trigger2
** (process:3460): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.4.auto/trigger3
** (process:3460): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3460): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
^C
$ systemctl start iio-sensor-proxy.service
$ monitor-sensor
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
    Accelerometer orientation changed: normal
    Light changed: 8.000000 (lux)
    Light changed: 9.000000 (lux)
    Light changed: 13.000000 (lux)
    Accelerometer orientation changed: left-up
    Light changed: 78.000000 (lux)
    Light changed: 58.000000 (lux)
    Accelerometer orientation changed: normal
    Light changed: 30.000000 (lux)
    Light changed: 8.000000 (lux)
```

hadess, 2016-06-19:
iio-sensor-proxy and the device are reporting brightness changes correctly. From the screenshots, I see you're not using GNOME either. You'd need to ask Ubuntu/Canonical why there's no integration or why that integration doesn't work, because they don't use gnome-settings-daemon.

closed 2016-06-19 by hadess