hadess/iio-sensor-proxy #82: iio-sensor-proxy service starts too early causing screen rotation to fail

ignatenkobrain, 2016-05-05:
Description of problem:
On some platform, I have noticed that rotation doesn't work after poweron but starts working after suspend/resume of the device.

Version-Release number of selected component (if applicable):

How reproducible:
100% on some device like Yoga 900

Steps to Reproduce:
1. Powerup device with HID sensor hub (For example Yoga 900).
2. After GUI starts try to rotate device.

Actual results:
Screen rotation doesn't work

Expected results:
Screen rotation should work once user changes the device orientation

Additional info:
Works after suspend and then resume

ignatenkobrain, 2016-05-05:
This service starts too early. I have added prints in the kernel, where request is issued to powerup the sensors. I don't see any request to power on sensors.

The possible solution is to add to
iio-sensor-proxy.service

```
[Unit]
After=multi-user.target
```

This will delay enough to make the linux-iio ready to accept requests.

ignatenkobrain, 2016-05-05:
Originally reported: https://bugzilla.redhat.com/show_bug.cgi?id=1333541

hadess, 2016-05-05:
Fairly certain this is a driver bug, and nothing to do with iio-sensor-proxy itself.

spandruvada, 2016-05-09:
Not sure. what do you mean by driver bug here. If the sensor proxy starts before IIO modules are loaded and initialized, then no one will be there to process requests.

hadess, 2016-05-10:
As far as I understood, and maybe the docs fail me here, but After= just defines a dependency ordering, not an actual dependency.

In any case, we can probably remove that line, as iio-sensor-proxy will be started as needed by udev.

Espionage724, 2016-05-22:
Can also confirm this on my HP Pavilion x2. With Fedora 23 I believe back on kernel 4.2, iio-sensor-proxy worked without any intervention (screen auto-brightness and rotation). Currently, iio-sensor-proxy only seems to work without intervention from a LiveCD, and after a suspend on an actual install.

Adding After=multi-user.target to the service makes iio-sensor-proxy work fine again.

hadess, 2016-05-25:
@ignatenkobrain seems like it could be a race in iio-sensor-proxy.

What I think happens is:
1) iio-sensor-proxy gets activated on boot, before the iio driver is loaded
2) we set up dbus and get the list of devices from udev
3) device appears, after the initial probe
4) iio-sensor proxy exits because we couldn't find anything in 2)

Any chance you could add debug to iio-sensor-proxy so we can root that out?

andreacampanella, 2016-05-25:
Can iio-sensor-proxy check lsmod to check if the module is loaded? As a crude hack this should work...

ignatenkobrain, 2016-05-25:
@andreacampanella no, this is ugly and will never happen.

ignatenkobrain, 2016-05-25:
@hadess it's better to ask original reported from RHBZ (I don't have such devices)

ignatenkobrain, 2016-05-25:
actually, @spandruvada is original reporter.

fajabird, 2016-05-25:
Seems to be the same as: https://github.com/hadess/iio-sensor-proxy/issues/66

blkeller, 2016-06-06:
I can also confirm this problem:
- Dell Inspiron 11 - 3000 series laptop (model 3147)
- Ubuntu GNOME 16.04 64-bit
- GNOME Shell 3.18.4
- Linux kernel 4.4.0-22-generic
- iio-sensor-proxy 1.1-1

I thought I was loosing my mind because screen rotation was working perfectly since initial install.  Then I upgraded from a platter-based hard drive to an SSD, and screen rotation suddenly broke.  My boot times shortened by about 80%, and based on this bug report, I guess the boot process is now too fast and I'm hitting this bug.

The `After=multi-user.target` workaround takes care of the problem only about 60-70% of every reboot for me.  I have to resort to a suspend/resume sequence to bring back rotation the other 40-30% of the time.

ctubbsii, 2016-08-29:
@kekx workaround on #85 worked for me.
I changed `Type=dbus` to `Type=idle` and all is good.

jeppebundsgaard, 2016-10-21:
I am on a Yoga 900. Ubuntu Gnome 16.10.
This one works for me: `systemctl restart iio-sensor-proxy`
Neither change of `Type=dbus` to `Type=idle` nor `[Unit]
After=multi-user.target` changed anything. 
Wonderful to have automatic rotation! Would be nice, though, not to have to restart the sensor proxy on each reboot.

dinojr, 2016-10-28:
On a Surface Pro, the `Type=idle` trick is working so far.

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

FWIW, despite what the title says, it starts at the right time (as early as possible), but the drivers below aren't initialised properly, or fail to report an error.

closed 2016-11-22 by hadess

spandruvada, 2017-08-10:
I posted a patch "[RFT][PATCH] iio: hid-sensor-trigger: Fix the race with user space powering up sensors. It will be great if anybody can test by taking out the workaround. This way I know that problem is fixed for good.

keeferrourke, 2017-08-18:
@spandruvada the race condition is still present on my Yoga 710, which boots *very* quickly on Fedora Workstation 26. How can I help test your patch?