hadess/iio-sensor-proxy #83: Lenovo Yoga 300: Could not find any supported sensors

sils1297, 2016-05-06:
Hi,

I'm getting

```
[root@gmob lasse]# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:1525): DEBUG: Could not find any supported sensors
```

With

```
[root@gmob lasse]# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 5986:0671 Acer, Inc 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Any hint on how I can get the appropriate driver to make this work or where to ask for more information?

hadess, 2016-05-06:
Decompile the DSDT and look for the accelerometer device in the list. If you find one, google for more details about the device, and check whether Linux has a driver for it. If it does, ask your distribution to enable it.

sils1297, 2016-05-19:
Hm, if I grep through it I can't find anything that looks like the accelerometer device. (Though it's definitely there, it works on windows.)

hadess, 2016-05-19:
Put the HTML output of WinAudit available somewhere, and I can take a look.

sils1297, 2016-05-19:
Thank you so much, you've got mail.

hadess, 2016-05-19:
From there, it says that the accelerometer is from "Bitland Corporation" with the APCI ID "DUAL250E". I have no idea what's necessary to get it working, but it needs at least a driver for iio-sensor-proxy to do something.

Your best bet is to send a mail to the linux-iio mailing-list and ask if anybody knows about this device, or whether specifications are available.

Closing, as this is a kernel problem.

closed 2016-05-19 by hadess

sils1297, 2016-05-19:
Thanks very much for your help.