hadess/iio-sensor-proxy #84: Toshiba radius 12 sensor in sysfs but no updates in dbus

lJoublanc, 2016-05-17:
On the abovementioned laptop, I have four devices under `/sys/bus/iio`. If I `cat` the accelerometer inclination files, I can see the value changing as I modify the tilt of the laptop. However, when running iio-sensor-proxy with debug enabled, it doesn't appear to receive updates...

```
** (process:3576): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8350.0001/HID-SENSOR-200073.2.auto/iio:device3
** (process:3576): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8350.0001/HID-SENSOR-200073.2.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:3576): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:048D:8350.0001/HID-SENSOR-200073.2.auto/trigger3
** (process:3576): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3576): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3576): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3576): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3576): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3576): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:3576): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:3576): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:3576): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3576): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3576): DEBUG: No new data available
** (process:3576): DEBUG: No new data available
** (process:3576): DEBUG: No new data available
** (process:3576): DEBUG: No new data available
** (process:3576): DEBUG: No new data available
...
```

I'm running kernel 4.4.7.

Also, would appreciate if anyone can help me find out what method I can use to figure out what sensors I have - currently lsmod doesn't appear to show any specific modules; just generic `hid_sensor_...`. I've built all the accelerometer modules under the sun, so I'd like to disable the ones I'm not using.

lJoublanc, 2016-05-22:
I've found that , like in a lot of other threads, this appears to be resolved after waking up from a soft suspend. I'll try updating my kernel at some point.

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

closed 2016-11-22 by hadess