hadess/iio-sensor-proxy #85: Unable to determine accelerator orientation

bgaff, 2016-05-19:
For some reason screen rotation stopped working suddenly for me, if I query the devices via: `watch -n 0.1 cat /sys/bus/iio/devices/iio\:device*/*raw* /dev/iio*` I will see the devices change with my rotation. I tried rebuilding iio-sensor-proxy from master and still it's unable to determine the orientation.

`$ uname -r
4.4.0-22-generic
`

`
$ sudo gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:ien
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};
`

bgaff, 2016-05-19:
The output from running iio-proxy-sensor directly via GDB:

```
(gdb) r
Starting program: /usr/local/iio-sensor-proxy-master/sbin/iio-sensor-proxy 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff5912700 (LWP 11689)]
[New Thread 0x7ffff5111700 (LWP 11690)]
[New Thread 0x7ffff4910700 (LWP 11691)]
** (process:11685): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0002/HID-SENSOR-200041.7.auto/iio:device3
** (process:11685): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0002/HID-SENSOR-200041.7.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
** (process:11685): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0002/HID-SENSOR-200073.2.auto/iio:device2
** (process:11685): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0002/HID-SENSOR-200073.2.auto/iio:device2 of type accelerometer at IIO Buffer accelerometer
** (process:11685): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0002/HID-SENSOR-200083.4.auto/iio:device1
** (process:11685): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0002/HID-SENSOR-200083.4.auto/iio:device1 of type compass at IIO Buffer Compass
** (process:11685): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0002/HID-SENSOR-200073.2.auto/trigger2
** (process:11685): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Built channel array for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Built channel array for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Built channel array for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0002/HID-SENSOR-200041.7.auto/trigger3
** (process:11685): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:91D1.0002/HID-SENSOR-200083.4.auto/trigger1
** (process:11685): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:11685): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:11685): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:11685): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:11685): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:11685): DEBUG: No new data available
** (process:11685): DEBUG: No new data available
** (process:11685): DEBUG: No new data available
```

bgaff, 2016-05-19:
Like I said, this was working about a day ago, nothing changed (as far as I can tell) on my system. I even tried the only other kernel version available on my system and it had the same problem. Any suggestions on what/where to check to figure out what could have changed or broke?

bgaff, 2016-05-19:
After a little more digging I see that people suggested that things will work after a suspend, sure enough I tried suspending and now it's able to determine orientation. This seems to suggest a kernel bug.

```
$ sudo gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'normal';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 71.0;
  };
  node Compass {
  };
};
```

nacc, 2016-07-04:
@bgaff I'm still seeing this issue (Ubuntu 16.04). S/R seems to make no difference (although I believe it used to...). Are you still seeing successful rotation with current kernels?

nacc, 2016-07-05:
@bgaff FYI, I just built and booted into 4.7-rc5+ (0b295dd5b8eb755bc7b57220d93705994090e954 is my current tip HEAD) and screen rotation worked fine, regardless of suspend & resume.

kekx, 2016-07-23:
@nacc Hm, I just installed the 4.7-rc7 kernel from rawhide and it still doesn't work without suspend & resume. Using a Surface Pro 2 btw which is tested according to README.

What might be interesting for @hadess though is that after S&R and while running iio-server-proxy with debug output, as soon as I clip on my type cover, iio-server-proxy seems to lose all the sensors and I have to S&R again to regain functionality. Maybe attaching devices mixes up the iio devices or something.

kekx, 2016-07-24:
Found a workaround for now: I simply added Type=idle to the service file to make it start later in the boot process. It never worked before directly after boot, now it does! And even after disconnecting the type cover, while it still stops working, it caught up again after maybe a minute and worked again.

bgaff, 2016-07-24:
Yah on 4.7-rc5 it still doesn't properly work, can you either provide more details?

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

closed 2016-11-22 by hadess