hadess/iio-sensor-proxy #88: Screen blank when tilted 90 degrees

lJoublanc, 2016-05-22:
I've installed iio-sensor-proxy, and it seems to work now after waking from suspend (see my other post #84).

But now I have another (unrelated?) issue. When the screen is tilted at 90 degrees, it goes blank. When I put it upright again, I appear to see the rotated screen flash by for an instant until the upright screen comes back up.

Rotating 180 degrees works fine.

I don't know if this is related, but I've also noticed that I don't have an option for orienting my screen in the gnome display settings, as I do for example on my desktop computer. Could this be somehow related?

hadess, 2016-05-22:
> But now I have another (unrelated?) issue. When the screen is tilted at 90 degrees, it goes blank.  When I put it upright again, I appear to see the rotated screen flash by for an instant until the upright screen comes back up.

> Rotating 180 degrees works fine.

That would be a bug in the display driver. You can rest the tablet, and run `xrandr -o left` in a terminal. If the screen goes blank, you're now certain that the problem is driver related.

> I don't know if this is related, but I've also noticed that I don't have an option for orienting my screen in the gnome display settings, as I do for example on my desktop computer. Could this be somehow related?

There's no rotation buttons when there's automatic rotation through an accelerometer, that's expected.

lJoublanc, 2016-05-22:
Thanks; I tried to change the orientation manually and you're correct, the screen is blank when I do this.

Also I found a number of issues googling, looks like it may be related to kernel bug [109501](https://bugzilla.kernel.org/show_bug.cgi?id=109501)

hadess, 2016-05-23:
> Thanks; I tried to change the orientation manually and you're correct, the screen is blank when I do this.

Than it's a graphics driver issue.

> Also I found a number of issues googling, looks like it may be related to kernel bug 109501

That's not a graphics driver issue, it's an issue with the accelerometer. If the display is blank when you rotate the configuration, then it has nothing to do with the accelerometer.

In any case, kernel bugs.

closed 2016-05-23 by hadess