hadess/iio-sensor-proxy #90: Hold off on actually rotating screen or changing brightness for a second or two

badshah400, 2016-06-10:
On my system iio-sensor-proxy detects the accelerometer and ambient brightness and works great. My only complaint is that it works too well really, i.e. it is somewhat over sensitive to orientation and/or brightness changes. It so happens that if you quickly tilt the laptop and switch it back, the display rotates and remains stuck in the tilted orientation. You have to tilt and switch back again to actually get the display back to the original orientation. I suggest holding off on actually rotating the display until the laptop remains in the new orientation for at least 1 or 1.5 sec.

Ditto for ambient brightness changes. The display flickering because of small brightness changes in a dimply lit room is a little bit troubling on the eyes.

Thanks for your great application though. Has really made working with convertible laptops easy.

hadess, 2016-09-06:
iio-sensor-proxy isn't the one applying changes for brightness, and doesn't do anything other than report them. So you'd need to report that problem to your desktop environment.

As for the orientation/acceleration, I'd be happy to take patches to make the orientation changes smoother (iio-sensor-proxy doesn't apply the screen rotation either, for what it's worth).

janbrummer, 2017-03-25:
> On Sat, 2017-03-25 at 18:55 +0100, Jan-Michael Brummer wrote:
> 
>  Am Sa, 25. Mär, 2017 um 5:53 schrieb Bastien Nocera <hadess@hadess.ne
>  t>:
>  > On Sat, 2017-03-25 at 13:13 +0100, Jan-Michael Brummer wrote:
>  >  Hi Bastien,
>  >  
>  >  just stumbled on this issue myself: https://github.com/hadess/iio-
>  > sen
>  >  sor-proxy/issues/90
>  >  
>  >  Here is a patch for this issue.
>  > 
>  > I'm pretty sure that doesn't "make the orientation changes
>  > smoother",
>  > it just delays them.
>  
>  No one said that it would be smoother. The bug report says that the
>  trigger for the change is too frequent and therefore it should be
>  delayed (1 to 1,5s).
>  This removes unnecessary changes due to minimal changes in the hand
>  or walking
>  with the tablet. And this is what the patch achieves.
>  
>  > A better fix would be to use the gyro as well as the accelerometer,
>  > and/or use triggers to only get woken up with larger changes,
>  > removing
>  > the need to poll (so we get to know straight away when the value
>  > changed, not only on the polling timeout).
>  
>  This might work, but is a little more complex and wouldn't work on
>  device without
>  a gyro, that's why i'm think the timer version suits better.
>  
>  What do you think?
> 
> I'd have to test it. Can you create either a pull request, or put the
> contents of this discussion in the issue, and link to the patch
> somewhere?
> 
> I'll look into it once I've fixed an IIO kernel bug, and switch to
> working on iio-sensor-proxy again.
> 
> Cheers

Patch: https://www.tabos.de/download/0001-accel-Hold-off-on-actually-rotating-screen.patch

cas--, 2018-12-03:
Can this be revisited please. The link to the suggested patch is no longer available. 

I have a Lenovo X1 Yoga and find that brief slight tilts will switch screen orientation. Also when rotating the screen from `normal` to `bottom-up` the sensor sees `right-up` or `left-up` orientations which is quite undesired as the desktop takes several seconds to render those changes before finally settling on the correct orientation.

Device drivers are not really my area of expertise but I did a bit of research and found that perhaps a 1-2s delay using a workqueue function e.g. `scheduled_delayed_work` to call `send_dbus_event` in `accel_changed_func` and cancel the workqueue if further orientation calls are made before the event is called. Would this be the right track?

hadess, 2018-12-03:
> Device drivers are not really my area of expertise but I did a bit of research and found that perhaps a 1-2s delay using a workqueue function e.g. `scheduled_delayed_work` to call `send_dbus_event` [...]

I don't think that trying to use kernel API in user-space will work. iio-sensor-proxy is a user-space application, taking data from the driver, the functions you mention are kernel APIs available to kernel drivers. That's just not going to work.

cas--, 2018-12-03:
Ah ok, so perhaps [libevent](http://libevent.org/)? Do you have an idea on how to solve this?

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/90

closed 2020-01-09 by hadess