hadess/iio-sensor-proxy #91: iio-poll-accel driver hardcoded device name

Icenowy, 2016-06-17:
It supports only iio devices named "i2c-SMO8500:00".
But now I have an Allwinner tablet with a mma8653 polling accelerometer. It's configured at the device tree, and will have the name "mma8653" given by the driver.

I did an ugly hack on it ( replace "i2c-SMO8500:00" to "mma8653" in iio_poll_accel_discover() ), now iio-sensor-proxy works (although orientation is 90 degrees off)

hadess, 2016-09-06:
Ideally, the ->discover in `drv-iio-poll-accel.c` would use the capabilities (eg. the files in the sysfs directory) of the device instead of hard-coded names to figure out whether an accelerometer is supported. Want to look into it?

hadess, 2016-09-06:
See also #59

hadess, 2016-11-23:
This is actually a duplicate of #39

closed 2016-11-23 by hadess

hadess, 2016-11-23:
(which is now fixed in master)