hadess/iio-sensor-proxy #92: Wrong

M0Rf30, 2016-06-20:
Hi,
I've tested this piece of software on an asus tp500lb
Accelerometer is correctly detected, but on service start the screen is oriented in the wrong way.
In landscape the screen is inverted. Module is a kionix kxcjk-1013

hadess, 2016-06-20:
Does the orientation change between grub and the graphical bootloader?

M0Rf30, 2016-06-20:
No, Efi grub is loaded correctly with no graphical issues.

M0Rf30, 2016-06-21:
no, the linux graphical boot is not inverted. however in landscape  mode sensor-monitor shows bottom-up, in portrait mode same issue

M0Rf30, 2016-06-21:
here some details:
P: /devices/pci0000:00/INT3432:00/i2c-5/i2c-SMO8500:00/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/pci0000:00/INT3432:00/i2c-5/i2c-SMO8500:00/iio:device0
E: DEVTYPE=iio_device
E: MAJOR=245
E: MINOR=0
E: SUBSYSTEM=iio
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=10432122

M0Rf30, 2016-06-29:
any news on this?

hadess, 2016-09-06:
Your hardware uses the same accelerometer as the Onda v975w which needs a quirk. See `poll_orientation()` in `drv-iio-poll-accel.c`.

I don't know of an easy way to distinguish your tablet from this Onda one. Can you please attach the output of `dmidecode` to this bug?

M0Rf30, 2016-09-07:
dmidecode out
https://gist.github.com/M0Rf30/c1c22ef00948307f49e0c1dfe6c0ad12

hadess, 2016-09-09:
> you could extract the product name from dmi data and do a case switch, but this is only a humble suggestion for a person involved in such a project

Why do you think I asked for the dmidecode output? :)

hadess, 2016-11-23:
One more question, do you have the DSDT for that device somewhere?
I'd need the `dsdt.dsl` generated from those instructions (just steps 1 and 2):
https://wiki.archlinux.org/index.php/DSDT#Recompiling_it_yourself

M0Rf30, 2016-11-23:
I will provide you the dsdt as soon as possible. Maybe late in the afternoon

M0Rf30, 2016-11-23:
https://gist.github.com/M0Rf30/005f91a7603311df19b6ef17da0cba17

hadess, 2016-11-23:
OK, there's no _PLD so we'll have to use a quirk in udev for that device.

hadess, 2016-12-12:
You can add a quirk for this in systemd/udev. See the "Accelerometer orientation" of the README.

closed 2016-12-12 by hadess

M0Rf30, 2016-12-12:
I will test the quirk and do a pull request in systemd 60-sensor.hwdb file. thanks man

AlexGluck, 2016-12-30:
For asus tp300lj this not work, anyone can help me? 
cat /etc/udev/hwdb.d/61-sensor-local.hwdb
`#########################################
# AsusTek TP300LJ
#########################################
sensor:modalias:acpi:SMO8500*:dmi:*svn*ASUSTeK*:*pn*TP300LJ*
# ACCEL_MOUNT_MATRIX=0, -1, 0; -1, 0, 0; 0, 0, 0
`

hadess, 2016-12-30:
@AlexGluck wrong place to ask, the detection happens in systemd, so you should ask there.

But this is likely wrong:
```
ACCEL_MOUNT_MATRIX=0, -1, 0; -1, 0, 0; 0, 0, 0
```

The Z coordinate is just thrown away as the vector is all zeroes.

AlexGluck, 2016-12-30:
I downgrade version of iio-proxy and all works fine. I add vector for Z coordinate, but nothing happen. I think fix for driver kxcj-9 broke correctly work. After that you say, what i need ask in systemd.

M0Rf30, 2017-01-19:
it works with https://github.com/M0Rf30/scripts/blob/master/60-sensor.hwdb

AlexGluck, 2017-01-19:
i don't understand why, but it not working. I try replace TP500LB to TP300LJ. If it help i can post dmidecode out.

M0Rf30, 2017-01-19:
Are you using systemd-git? Accel_mount_matrix support is for latest commits of systemd and linux kernel>4.6

AlexGluck, 2017-01-19:
No, i use fedora 25 with latest updates, systemd version now 231. Kernel 4.9.3. 
```
[alexgluck@gluck-laptop]$ lsiio
Device 001: i2c-SMO8500:00
Device 000: acpi-als
```
How i can check matrix uses for me?

M0Rf30, 2017-01-19:
if you install latest systemd-git, you will find the accel_mount _matrix using the udevadm command listed in the README.md of iio-sensor-proxy. Execute the first step and I'll try to help you

M0Rf30, 2017-01-19:
Done
https://github.com/systemd/systemd/pull/5111