hadess/iio-sensor-proxy #93: ColorHugALS not detected in lenovo x220

jjardon, 2016-07-02:
Hi,

using Arch updated as today

ColorHugALS (http://www.hughski.com/colorhugals.html) doesnt seems to be detected:

$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:5679): DEBUG: Could not find any supported sensors

If I manually use the colorhug-backlight calibration tool the sensors work fine

dmes message when connecting the sensor:
[  613.412603] usb 2-1.1: new full-speed USB device number 5 using ehci-pci
[  613.501902] hid-generic 0003:273F:1007.0003: hiddev0,hidraw0: USB HID v1.11 Device [Hughski Ltd. ColorHugALS] on usb-0000:00:1d.0-1.1/input0

jjardon, 2016-07-02:
P: /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1
N: bus/usb/002/006
E: BUSNUM=002
E: COLORD_IGNORE=1
E: COLORD_SENSOR_KIND=colorhug-als
E: COLOR_MEASUREMENT_DEVICE=1
E: DEVNAME=/dev/bus/usb/002/006
E: DEVNUM=006
E: DEVPATH=/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1
E: DEVTYPE=usb_device
E: DRIVER=usb
E: ID_BUS=usb
E: ID_FOR_SEAT=usb-pci-0000_00_1d_0-usb-0_1_1
E: ID_MODEL=ColorHugALS
E: ID_MODEL_ENC=ColorHugALS
E: ID_MODEL_ID=1007
E: ID_PATH=pci-0000:00:1d.0-usb-0:1.1
E: ID_PATH_TAG=pci-0000_00_1d_0-usb-0_1_1
E: ID_REVISION=0002
E: ID_SERIAL=Hughski_Ltd._ColorHugALS
E: ID_USB_INTERFACES=:030000:
E: ID_VENDOR=Hughski_Ltd.
E: ID_VENDOR_ENC=Hughski\x20Ltd.
E: ID_VENDOR_ID=273f
E: MAJOR=189
E: MINOR=133
E: PRODUCT=273f/1007/2
E: SUBSYSTEM=usb
E: TAGS=:uaccess:seat:
E: TYPE=0/0/0
E: USEC_INITIALIZED=2827211530

and

P: /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:273F:1007.0004
E: DEVPATH=/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:273F:1007.0004
E: DRIVER=hid-generic
E: HID_ID=0003:0000273F:00001007
E: HID_NAME=Hughski Ltd. ColorHugALS
E: HID_PHYS=usb-0000:00:1d.0-1.1/input0
E: HID_UNIQ=
MODALIAS=hid:b0003g0001v0000273Fp00001007
E: SUBSYSTEM=hid

hadess, 2016-07-02:
Not sure which distribution you're using, but it doesn't show up as an iio device, which it should. Missing IIO and HID sensor hub support?

jjardon, 2016-07-02:
I'm using Archlinux

>   Missing IIO and HID sensor hub support?

Does this need to be enabled in the kernel? If yes, this is the current default config: https://git.archlinux.org/svntogit/packages.git/tree/trunk/config.x86_64?h=packages/linux

hadess, 2016-07-02:
Did you update it to the latest firmware already? The older versions used a proprietary protocol (eg. a protocol only used by this device).

jjardon, 2016-07-02:
I've used the colorhug-flash utility and it reports Im using the latest firmware: 3.0.2, and seems its the latest available: http://www.hughski.com/downloads/colorhug-als/firmware/

Maybe I should use another way to switch to the open firmware?

hadess, 2016-07-02:
Latest version is 4.0.3:
https://secure-lvfs.rhcloud.com/lvfs/device/84f40464-9272-4ef7-9399-cd95f12da696

closed 2016-07-02 by hadess