hadess/iio-sensor-proxy #100: orientation undefined in 1.2

rickysarraf, 2016-09-12:
This seems to be a regression in 1.2. I'm reporting these issues when running on my 4.7.3 kernel.

With 1.2,  I see some odd problems.

On fresh boot:

```
rrs@learner:~$ monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
--- iio-sensor-proxy vanished, waiting for it to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
+++ Accelerometer appeared
    Accelerometer orientation changed: undefined
+++ Light sensor appeared
    Light changed: 0.000000 (lux)
^C
2016-09-12 / 21:55:03 ♒♒♒  ☹  => 130  
```

Then, on 1st suspend/resume, which is a workaround in such cases, only 360 tilt works.

```
rrs@learner:~$ monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: normal)
=== Has ambient light sensor (value: 39.999999, unit: lux)
    Light changed: 10.000000 (lux)
    Light changed: 39.999999 (lux)
    Accelerometer orientation changed: bottom-up
    Light changed: 0.000000 (lux)
    Accelerometer orientation changed: normal
    Light changed: 39.999999 (lux)
    Light changed: 10.000000 (lux)
    Light changed: 39.999999 (lux)
    Light changed: 10.000000 (lux)
    Accelerometer orientation changed: bottom-up
    Accelerometer orientation changed: normal
    Light changed: 39.999999 (lux)
    Light changed: 29.999999 (lux)

```

rickysarraf, 2016-09-12:
On 4.8 kernel, things go much screwed up.

As you see, on fresh boot, no orientation details are available. Upon suspend/resume, it gets further screwed up. Instead of normal orientation, it takes right/left orientation as a default. And 360 tilt changes orientation to right or left only, effectively leaving the 1.2 release not very useful, irrespective of the kernel in use.

```
rrs@learner:~$ monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)

    Accelerometer orientation changed: right-up
    Light changed: 10.000000 (lux)
    Light changed: 49.999999 (lux)
    Light changed: 20.000000 (lux)
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: right-up

^C2016-09-12 / 22:06:15 ♒♒♒  ☹  => 130  
rrs@learner:~$ 
```

rickysarraf, 2016-09-12:
Reverting back to 1.1 solved the problem.

rickysarraf, 2016-09-12:
Here's some debug data with 1.2 running atop 4.7.3 kernel

```
rrs@learner:/var/tmp/Debian-Build/Result$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:10579): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3
** (process:10579): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
** (process:10579): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device0
** (process:10579): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:10579): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device4
** (process:10579): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device4 of type compass at IIO Buffer Compass
** (process:10579): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto/trigger0
** (process:10579): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:10579): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:10579): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:10579): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:10579): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:10579): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:10579): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200041.9.auto/trigger3
** (process:10579): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10579): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10579): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200083.6.auto/trigger4
** (process:10579): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10579): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10579): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10579): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:10579): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:10579): DEBUG: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10579): DEBUG: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10579): DEBUG: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10579): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:10579): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:10579): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:10579): DEBUG: process_scan_1: channel_index: 2, chan_name: in_accel_x, channel_data_index: 0 location: 8
** (process:10579): DEBUG: process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
** (process:10579): DEBUG: process_scan_1: channel_index: 0, chan_name: in_accel_z, channel_data_index: 2 location: 0
** (process:10579): DEBUG: Read from IIO on 'iio:device0': -230, -968, 19
** (process:10579): DEBUG: Accel sent by driver (quirk applied): 2, 9, 0
** (process:10579): DEBUG: Emitted orientation changed: from undefined to normal
** (process:10579): DEBUG: No new data available on 'iio:device3'
** (process:10579): DEBUG: process_scan_1: channel_index: 2, chan_name: in_accel_x, channel_data_index: 0 location: 8
** (process:10579): DEBUG: process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
** (process:10579): DEBUG: process_scan_1: channel_index: 0, chan_name: in_accel_z, channel_data_index: 2 location: 0
** (process:10579): DEBUG: Read from IIO on 'iio:device0': -230, -968, 23
** (process:10579): DEBUG: Accel sent by driver (quirk applied): 2, 9, 0
** (process:10579): DEBUG: No new data available on 'iio:device3'
** (process:10579): DEBUG: No new data available on 'iio:device0'
[... cut ...]
** (process:10579): DEBUG: process_scan_1: channel_index: 2, chan_name: in_accel_x, channel_data_index: 0 location: 8
** (process:10579): DEBUG: process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
** (process:10579): DEBUG: process_scan_1: channel_index: 0, chan_name: in_accel_z, channel_data_index: 2 location: 0
** (process:10579): DEBUG: Read from IIO on 'iio:device0': -218, -968, 27
** (process:10579): DEBUG: Accel sent by driver (quirk applied): 2, 9, 0
** (process:10579): DEBUG: No new data available on 'iio:device3'
^C
2016-09-12 / 22:27:27 ♒♒♒  ☹  => 130  
```

hadess, 2016-09-12:
> ** (process:10579): DEBUG: Read from IIO on 'iio:device0': -218, -964, 27
> ** (process:10579): DEBUG: Accel sent by driver (quirk applied): 2, 9, 0

Can you add debug in `src/drv-iio-buffer-accel.c`'s `process_scan()` function, to print out the scale?

Something like:
```
g_debug ("Accel read from IIO on '%s': %d, %d, %d (scale %lf)", or_data->name, accel_x, accel_y, accel_z, scale);
```

Looks like the scale might be broken on your device.

rickysarraf, 2016-09-12:
Here's the data you asked for.

```
rrs@learner:/media/SSHD/var/tmp/Debian-Build/Result$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:30484): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3
** (process:30484): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200041.9.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
** (process:30484): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device0
** (process:30484): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:30484): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device4
** (process:30484): DEBUG: Found device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200083.6.auto/iio:device4 of type compass at IIO Buffer Compass
** (process:30484): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto/trigger0
** (process:30484): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:30484): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:30484): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:30484): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:30484): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:30484): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:30484): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200041.9.auto/trigger3
** (process:30484): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:30484): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:30484): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200083.6.auto/trigger4
** (process:30484): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:30484): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:30484): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:30484): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:30484): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:30484): DEBUG: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:30484): DEBUG: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:30484): DEBUG: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:30484): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:30484): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:30484): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:30484): DEBUG: No new data available on 'iio:device3'
** (process:30484): DEBUG: process_scan_1: channel_index: 1, chan_name: in_accel_x, channel_data_index: 0 location: 4
** (process:30484): DEBUG: process_scan_1: channel_index: 0, chan_name: in_accel_y, channel_data_index: 1 location: 0
** (process:30484): DEBUG: process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
** (process:30484): DEBUG: Read from IIO on 'iio:device0': -910, 3, -378
** (process:30484): DEBUG: Accel sent by driver (quirk applied): 8, 0, -3
** (process:30484): DEBUG: Emitted orientation changed: from undefined to right-up
** (process:30484): DEBUG: Accel read from IIO on 'iio:device0': 910, -3, -378 (scale 0.009806)
** (process:30484): DEBUG: No new data available on 'iio:device3'
** (process:30484): DEBUG: process_scan_1: channel_index: 1, chan_name: in_accel_x, channel_data_index: 0 location: 4
** (process:30484): DEBUG: process_scan_1: channel_index: 0, chan_name: in_accel_y, channel_data_index: 1 location: 0
** (process:30484): DEBUG: process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
** (process:30484): DEBUG: Read from IIO on 'iio:device0': -906, 7, -441
** (process:30484): DEBUG: Accel sent by driver (quirk applied): 8, 0, -4
[... cut ...]
** (process:30484): DEBUG: process_scan_1: channel_index: 0, chan_name: in_accel_y, channel_data_index: 1 location: 0
** (process:30484): DEBUG: process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
** (process:30484): DEBUG: Read from IIO on 'iio:device0': -921, 3, -402
** (process:30484): DEBUG: Accel sent by driver (quirk applied): 9, 0, -3
** (process:30484): DEBUG: Accel read from IIO on 'iio:device0': 921, -3, -402 (scale 0.009806)
^C
2016-09-13 / 00:07:40 ♒♒♒  ☹  => 130  
```

hadess, 2016-09-13:
There should be a few "scale" files under:
`/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device0`

Can you run the following command:
```
find /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto/iio:device0 -iname "*scale*" -print -exec cat '{}' \;
```

rickysarraf, 2016-09-13:
There is no iio:device0 but instead had iio:device2. Perhaps these names are persistent ones. Not sure.

But I anyways have extracted the output with relevant device:

```
rrs@learner:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto$ find iio\:device2/ -iname "*scale*" -print -exec cat '{}' \;
iio:device2/in_accel_scale
0.009806
2016-09-13 / 22:37:29 ♒♒♒  ☺  

rrs@learner:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:048D:8350.0002/HID-SENSOR-200073.4.auto$ ls
driver@  driver_override  iio:device2/  modalias  power/  subsystem@  trigger2/  uevent
2016-09-13 / 22:39:49 ♒♒♒  ☺  

```

hadess, 2016-09-13:
This sounds like a bug in the driver, or the HID report for that device. Can you please send a mail to the linux-iio mailing-list asking about the bizarre acceleration scaling? For your device, I would expect this to be closer to 0.5 or 0.25, given the unscaled values.

rickysarraf, 2016-09-13:
Okay! Thanks @hadess I'll do that. But can you please give me some information ?

As I understand from your comments, the scaling feature is added in 1.2 ?
Because after I revert to 1.1, things work fine for me.

hadess, 2016-09-13:
Yes, it was added to fix problems with other devices, where the scaling matters (commit 0975741edb62f6699277e1cda060fbcbb4ed5c20)

rickysarraf, 2016-09-13:
@hadess  There's one more issue on 1.2 with 4.8 kernel.

```
rrs@learner:~$ monitor-sensor     Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: right-up)
=== Has ambient light sensor (value: 59.999999, unit: lux)
    Accelerometer orientation changed: left-up
    Light changed: 0.000000 (lux)
    Accelerometer orientation changed: right-up
    Light changed: 59.999999 (lux)
    Light changed: 20.000000 (lux)
    Accelerometer orientation changed: left-up
    Accelerometer orientation changed: right-up
    Light changed: 49.999999 (lux)
--- iio-sensor-proxy vanished, waiting for it to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
    Accelerometer orientation changed: right-up
```

As you can see, there is no orientation called 'normal'. This issue, again, is only seen with 1.2.
Do you think this too is caused because of the driver ?

hadess, 2016-09-14:
I hope I got this right. Please test and reopen if there's a problem.

closed 2016-09-14 by hadess

rickysarraf, 2016-09-14:
Well. It is slightly better, as in now it can sense 'normal' orientation. But none of the orientation states are in line with the actual physical orientation of the device.

Logs attached.
[hadess.txt](https://github.com/hadess/iio-sensor-proxy/files/472090/hadess.txt)

hadess, 2016-09-14:
That would be what Srinivas mentioned on the mailing-list. It's a separate bug though.

hadess, 2016-09-14:
Please file a separate bug and try reverting 984803bc1b06e882bf3cf434fb4deb2ea3b98f65, and see whether that fixes it.

gevasiliou, 2016-09-14:
I am just following this bug , and i just got an idea that i would like to share with you;you might found it interesting. 
i was wondering if it would be easy & usefull to provide a kind of "callibration app" (i.e a a seperate file , called for example "proxy-sensor-cal")  

Imagine any user to be able to run this calibration app  like this: 
app prompt -> place your screen in normal position and then press enter.
app message -> "Setting stored".
app prompt -> place your screen in right up position and then press enter
app message -> "settings stored" or "error - no screen change detected - aborting".
etc

With such a solution you could match all screen models, set ups, all kernel versions, etc.

If this idea creates more problems than solutions, please ignore it. 
It was not my intentention to be the "smart ass" here.

Regards,
George

commit 07299fb5 (2016-09-14): accel: Fix accelerometer unit confusion

The original orientation code was based upon code that expected 1G to
roughly correspond to a reading of 256, to work-around the fact that the
input layer could not pass fractional values easily.

So the orientation_calc() code expected that too. But IIO readings after
scaling are properly in m/s², and our readings are integers.

We'll pass the scaling around to be applied at the last minute, when
doing calculations.

Closes: #100