hadess/iio-sensor-proxy #102: Tested on HP Pavilion X360

clefebvre, 2016-09-26:
Hi @hadess,

Congratulations on this great project. I'm happy to report that sensors and accelerometer orientation changes are successfully detected on the following test machine:

- HP Pavilion X360
- Kernel 4.4.0-21
- Linux Mint 18

clefebvre, 2016-09-26:
Also, support for orientation changes via iio-sensor-proxy will be present in Cinnamon, starting from version 3.2.

closed 2016-09-26 by hadess

Tiago-Marques, 2016-10-01:
@clefebvre Hi, i have a hp x360 too, ang can't get orientation work.

Ubuntu Gnome
4.4.0-38

Can you give me any help, please?

clefebvre, 2016-10-05:
Hi @Tiago-Marques,

The laptop is affected by the kernel issue, so it only works after a suspend-resume cycle.

Tiago-Marques, 2016-10-06:
@clefebvre You are right ! It's working now. Thank you ! Incredible job guys.

clefebvre, 2016-10-07:
Hi @Tiago-Marques,

Since you're using GNOME, can you test something quickly?

We fixed something in Cinnamon (https://github.com/linuxmint/cinnamon-settings-daemon/commit/b04d07b73695019034703abcd7d95c886706e214), but I'm not sure whether it needs fixing in GNOME (the settings daemon are slightly different).

To reproduce:

- Suspend/Resume to make the sensor work
- Rotate your laptop to the left (wait till the screen rotates).
- Rotate your laptop back into normal position (wait till the screen rotates back).
- Reboot

On the next reboot, because of the kernel bug, the sensors aren't active yet... check that GNOME starts with a normal orientation, and not rotated to the left.

If that's the case, then good news. Otherwise please let me know as we'll need to port that fix to GNOME.

NestorAcevedo, 2017-11-10:
Hi, in my HP Pavilion x360 with openSUSE Tumbleweed and KDE seems not work.

Kernel: 4.13.11-1-default

I already suspended/resumed my pc, rotated to left and normal position, rebooted but still does not work.

commit cbde8553 (2016-09-26): README: Add HP Pavilion X360

Closes #102