hadess/iio-sensor-proxy #105: iio-sensors not found - but present

modemaster, 2016-10-09:
Hi!
I have an ASUS TP200S with UBUNTU Gnome on it.
Kernel 
`Linux j-E205SA 4.4.0-38-generic #57-Ubuntu SMP Tue Sep 6 15:42:33 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

I can see the iio folders as follows:
`root@j-E205SA:/sys/bus/iio/devices/iio:device0# ls

buffer           in_accel_scale_available  in_anglvel_scale            in_anglvel_z_raw  in_temp_scale       sampling_frequency_available  uevent
dev              in_accel_x_raw            in_anglvel_scale_available  in_gyro_matrix    name                scan_elements
in_accel_matrix  in_accel_y_raw            in_anglvel_x_raw            in_temp_offset    power               subsystem
in_accel_scale   in_accel_z_raw            in_anglvel_y_raw            in_temp_raw       sampling_frequency  trigger`

udevadm info --export-db doesn't show iio devices.

As well as:
`root@j-E205SA:/sys/bus/iio/devices/iio:device0# systemctl stop iio-sensor-proxy.service
root@j-E205SA:/sys/bus/iio/devices/iio:device0# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy 
** (process:3526): DEBUG: Could not find any supported sensors
`

What can I do? With a script I can rotate the screen orientation through the gravity sensor. So in general it should work.

Thanks for helping!
Joel

ps: i posted recently a similar question. there i reported that i can see the iio sensors just on ubuntu-systems, not on fedora. I decided not to continue with fedora because it will be more headache...

hadess, 2016-11-22:
> udevadm info --export-db doesn't show iio devices

If it doesn't, then there shouldn't be one in /sys either. Attach the output here.

hadess, 2016-12-12:
Information requested 3 weeks ago and not given, closing. Reopen if you can provide the information. Would also be interesting if you could test against the current git master version.

closed 2016-12-12 by hadess