hadess/iio-sensor-proxy #106: Help needed on ASUS UX310UA running Arch

Kaukov, 2016-10-22:
Hey there!

Today I tried installing `iio-sensor-proxy`, but I failed to do so from the repo, so I downloaded the package from the AUR. Everything looks fine, but I can't enable the `iio-sensor-proxy.service`, I can only start it. 

    The unit files have no installation config (WantedBy, RequiredBy, Also, Alias
    settings in the [Install] section, and DefaultInstance for template units).
    This means they are not meant to be enabled using systemctl.
    Possible reasons for having this kind of units are:
    1) A unit may be statically enabled by being symlinked from another unit's
       .wants/ or .requires/ directory.
    2) A unit's purpose may be to act as a helper for some other unit which has
       a requirement dependency on it.
    3) A unit may be started when needed via activation (socket, path, timer,
       D-Bus, udev, scripted systemctl call, ...).
    4) In case of template units, the unit is meant to be enabled with some
       instance name specified.

I'm using Gnome as a DE, but when I start the service, Gnome doesn't have the option for automatic brightness. Please help!

hadess, 2016-10-24:
You're not supposed to start it by hand, it's started automatically when the sensor "appears". Reboot the machine and follow the instructions in the README. If iio-sensor-proxy isn't started, ask your distribution.

Kaukov, 2016-10-24:
Thanks, I asked on the forums!

closed 2016-10-24 by Kaukov